micro-manager/micro-manager #364: Images with large numbers of channels break the Image Inspector

kthorn, 2016-07-21:
I am opening a 13-channel image in Micro-manager 2.0 (the image was acquired in MM 1.4, if that is relevant). The image displays properly, but I cannot adjust the white and dark level sliders in the Image Inspector, and autostretch does not work. I am happy to send an example 13-channel image if it is helpful in debugging. I see the following error in the log file:

2016-07-21T14:59:18.896215 tid11328 [IFO,App] Error consuming images
                                    [       ] java.lang.ArrayIndexOutOfBoundsException: 7 in Thread[AWT-EventQueue-0,6,main]
                                    [       ]   at org.micromanager.display.internal.RememberedChannelSettings.saveSettingsToProfile(RememberedChannelSettings.java:227)
                                    [       ]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplaySettings(DefaultDisplayWindow.java:795)
                                    [       ]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplaySettings(DefaultDisplayWindow.java:780)
                                    [       ]   at org.micromanager.display.internal.LUTMaster.updateDisplayLUTForChannel(LUTMaster.java:323)
                                    [       ]   at org.micromanager.display.internal.LUTMaster.initializeDisplay(LUTMaster.java:266)
                                    [       ]   at org.micromanager.display.internal.CanvasUpdateQueue.showImage(CanvasUpdateQueue.java:325)
                                    [       ]   at org.micromanager.display.internal.CanvasUpdateQueue.consumeImages(CanvasUpdateQueue.java:232)
                                    [       ]   at org.micromanager.display.internal.CanvasUpdateQueue.access$000(CanvasUpdateQueue.java:81)
                                    [       ]   at org.micromanager.display.internal.CanvasUpdateQueue$1.run(CanvasUpdateQueue.java:147)
                                    [       ]   at java.awt.event.InvocationEvent.dispatch(Unknown Source)
                                    [       ]   at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
                                    [       ]   at java.awt.EventQueue.access$000(Unknown Source)
                                    [       ]   at java.awt.EventQueue$1.run(Unknown Source)
                                    [       ]   at java.awt.EventQueue$1.run(Unknown Source)
                                    [       ]   at java.security.AccessController.doPrivileged(Native Method)
                                    [       ]   at java.security.AccessControlContext$1.doIntersectionPrivilege(Unknown Source)
                                    [       ]   at java.awt.EventQueue.dispatchEvent(Unknown Source)
                                    [       ]   at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
                                    [       ]   at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
                                    [       ]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
                                    [       ]   at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
                                    [       ]   at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
                                    [       ]   at java.awt.EventDispatchThread.run(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App] Thread 15 (AWT-EventQueue-0) terminated with uncaught exception
2016-07-21T14:59:20.655215 tid11328 [IFO,App] java.lang.ArrayIndexOutOfBoundsException: 7
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at org.micromanager.display.internal.RememberedChannelSettings.saveSettingsToProfile(RememberedChannelSettings.java:227)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplaySettings(DefaultDisplayWindow.java:795)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplaySettings(DefaultDisplayWindow.java:780)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at org.micromanager.display.internal.inspector.HistogramsPanel$4.actionPerformed(HistogramsPanel.java:247)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at javax.swing.JToggleButton$ToggleButtonModel.setPressed(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at java.awt.Component.processMouseEvent(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at javax.swing.JComponent.processMouseEvent(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at java.awt.Component.processEvent(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at java.awt.Container.processEvent(Unknown Source)
2016-07-21T14:59:20.655215 tid11328 [IFO,App]   at java.awt.Component.dispatchEventImpl(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.Container.dispatchEventImpl(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.Component.dispatchEvent(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.Container.dispatchEventImpl(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.Window.dispatchEventImpl(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.Component.dispatchEvent(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue.access$000(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue$1.run(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue$1.run(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.security.AccessController.doPrivileged(Native Method)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.security.AccessControlContext$1.doIntersectionPrivilege(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.security.AccessControlContext$1.doIntersectionPrivilege(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue$2.run(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue$2.run(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.security.AccessController.doPrivileged(Native Method)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.security.AccessControlContext$1.doIntersectionPrivilege(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventQueue.dispatchEvent(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
2016-07-21T14:59:20.656215 tid11328 [IFO,App]   at java.awt.EventDispatchThread.run(Unknown Source)

ChrisWeisiger, 2016-07-21:
ImageJ does not support more than 7 channels in composite view mode. There is supposed to be an error dialog to this effect if you try to display a dataset with too many channels, but I haven't tested it recently so it may well be broken.

Does Color or Grayscale mode work?

kthorn, 2016-07-21:
>
> ImageJ does not support more than 7 channels in composite view mode. 
> There is supposed to be an error dialog to this effect if you try to 
> display a dataset with too many channels, but I haven't tested it 
> recently so it may well be broken.
>
> Does Color or Grayscale mode work?

It does throw an error message in composite mode. Grayscale works to 
display the image, but autostretch and the histogram sliders are still 
broken.

Kurt

ChrisWeisiger, 2016-07-22:
Okay, this is now fixed. Thanks for the report.

closed 2016-07-22 by ChrisWeisiger