micro-manager/micro-manager #372: mm2: Plugin code using internal packages should be commented

marktsuchida, 2016-07-26:
Our plugins (in plugins/ and libraries/) use a lot of `internal` packages of MMStudio. This is unavoidable due to the plugins being legacy code from before the cleaned-up 2.0 API.

But people look to existing plugins for coding examples.

So we should group the `import` statements importing `internal` packages and add a commend, something like

    // Imports for MMStudio internal packages
    // Plugins should not access internal packages, to ensure modularity and
    // maintainability. However, this plugin code is older than the current
    // MMStudio API, so it still uses internal classes and interfaces. New code
    // should not imitate this practice.

ChrisWeisiger, 2016-07-26:
Including a reference to the 2.0 Transition Guide ( https://micro-manager.org/wiki/Version_2.0_API_Transition_Guide ) in the comment would also be a good idea.

nicost, 2020-04-06:
Imports that I encountered while executing this task:

Duplicator:
import org.micromanager.internal.utils.MMDialog;
import org.micromanager.internal.utils.ProgressBar;

Example:
import org.micromanager.internal.utils.MMFrame;

HCS (CustomSettingsFrame, SBSPlate, SiteGenerator)
import org.micromanager.internal.utils.FileDialogs;
import org.micromanager.internal.utils.MMScriptException;
import org.micromanager.internal.utils.ReportingUtils;
import org.micromanager.internal.utils.NumberUtils;
import org.micromanager.internal.utils.MMFrame;
import org.micromanager.internal.utils.TextUtils;

MMClearVolumePlugin:
import org.micromanager.internal.utils.MMFrame;
and display.internal classes, which may be OK here
ScrollerPanel:

nicost, 2020-04-06:
Completed.  Lessons learned:
- There are still a few plugins that depend on ReportingUtils.  I removed all occurrences where it was obvious how to.  The remainder can be done, but needs a bit more thought.
- By far most dependencies on internal code are on classes in the internal.utils package. Especially:
 MMFrame
 MMDialog
 FileDialogs
 GUIUtils
 ProgressBar
 NumberUtils
 TextUtils
 JavaUtils
 MathFunctions
 AffineUtils
I do not think that it makes sense to tell plugin writers to not use these.  To the contrary, I want plugin writers to use these, since they result in a more uniform UI and avoid lots of duplicated code.  Now may be a good time to look at these and decide how to organize them.  Happy to help implement what @marktsuchida thinks is the best design.

nicost, 2020-04-06:
Commenting implemented in PR #757 
Will open a new issue to discuss use of the "internal.utils" package.

closed 2020-04-06 by nicost