micro-manager/micro-manager #378: mm2: StorageSinglePlaneTiffSeries can create unreadable datasets

ChrisWeisiger, 2016-07-27:
Use a script to generate images that have a position index in their `Coords`, but no position name in their `Metadata`. All of the image files will be placed into the "Pos0" directory, and you will then not be able to load the dataset.

marktsuchida, 2016-07-27:
Here's a repro case. Populate the position list, then run the script. The dataset has two positions, but both images go into the Pos0 subdirectory.
[singleplanetiff-repro.bsh.txt](https://github.com/micro-manager/micro-manager/files/387207/singleplanetiff-repro.bsh.txt)

ChrisWeisiger, 2016-07-28:
This has been "fixed", in the sense that you now cannot make a singleplane TIFF series dataset with multiple stage positions without specifying a `PositionName` metadata property for each image you save. If you try to save an image with no `PositionName` (or no metadata at all), you'll get an `IllegalArgumentException`.

A better long-term fix would be to generate ad-hoc position names based on the stage position index in the image's `Coords`. Such a fix will likely have to wait until the `StorageSinglePlaneTIFFSeries` class gets an overhaul, as it's rather a mess right now.

closed 2016-07-28 by ChrisWeisiger