micro-manager/micro-manager #379: On-the-Fly Processor Pipeline window always appears

nicost, 2016-07-27:
In this specific user profile, the On-The-Fly Processor Pipeline window appears upon every restart, even tough there are no processors in its list at all, and the window was closed before quitting the application.

nicost, 2016-07-27:
Actually, the behavior is even stranger.  The empty On-The-Fly processor window appears on startup.  Closing it, then opening the Flat-Field correction plugin makes the On-The-Fly Processor window appear (as expected), but it now has two rows, both with the Flat-Field Correction Processor selected.  This happen no matter what state these are in when quitting Micro-Manager.  This makes it difficult to trust what the application is doing with the acquired images.

marktsuchida, 2016-07-27:
Could you check what's in the `org.micromanager.internal.pipelineinterface.PipelineTableModel:saved pipeline configuration` entry in the user profile, after quitting µManager with an empty pipeline (and also after quitting µManager with the two rows that you describe)?
(Profiles are in `~/Library/Application Support/Micro-Manager` or `%LocalAppData%\Micro-Manager`.)

Also, is this on OS X or Windows, and are you using the standard launcher? (I'm asking because the behavior sounds like a potential duplicated singleton somewhere.)

nicost, 2016-07-28:
I was running this under Netbeans, but see the same behavior when running stand-alone.  This is on Windows.  Where is %LocalAppData% and how do I get to it?

marktsuchida, 2016-07-28:
In an Explorer window, you can click on the location bar and type `%LocalAppData%`. It's usually `\AppData\Local` in your home directory (but hidden by default).

nicost, 2016-07-28:
Thanks, great trick! 
After quitting with an empty pipeline it has:
```
"org.micromanager.internal.pipelineinterface.PipelineTableModel:saved pipeline configuration": {
    "PropVal": ["{\"isEnabledInLive\":true,\"pluginName\":\"org.micromanager.multichannelshading.MultiChannelShading\",\"name\":\"Flat-Field Correction\",\"configSettings\":\"{\\n  \\\"ChannelGroup\\\": {\\n    \\\"PropVal\\\": \\\"Channel\\\",\\n    \\\"PropType\\\": \\\"String\\\"\\n  },\\n  \\\"BackgroundFileName\\\": {\\n    \\\"PropVal\\\": \\\"E:\\\\\\\\Users\\\\\\\\MicroscopeCalibration\\\\\\\\Dark-FF\\\\\\\\MMStack.ome.tif\\\",\\n    \\\"PropType\\\": \\\"String\\\"\\n  },\\n  \\\"Presets\\\": {\\n    \\\"PropVal\\\": [\\n      \\\"Cy5\\\",\\n      \\\"Cherry\\\"\\n    ],\\n    \\\"PropType\\\": \\\"String array\\\"\\n  },\\n  \\\"PresetFiles\\\": {\\n    \\\"PropVal\\\": [\\n      \\\"E:\\\\\\\\Users\\\\\\\\Xiaowei\\\\\\\\FlatfieldImages\\\\\\\\Cy5_20x_From_Cells-B.tif\\\",\\n      \\\"E:\\\\\\\\Users\\\\\\\\Xiaowei\\\\\\\\FlatfieldImages\\\\\\\\Cherry_20x_WideField.tif\\\"\\n    ],\\n    \\\"PropType\\\": \\\"String array\\\"\\n  }\\n}\",\"isEnabled\":true}"],
    "PropType": "String array"
  },
```

After quitting with the two rows both containing the same flatfield correction plugin, it looks like:
```
"org.micromanager.internal.pipelineinterface.PipelineTableModel:saved pipeline configuration": {
    "PropVal": ["{\"isEnabledInLive\":true,\"pluginName\":\"org.micromanager.multichannelshading.MultiChannelShading\",\"name\":\"Flat-Field Correction\",\"configSettings\":\"{\\n  \\\"ChannelGroup\\\": {\\n    \\\"PropVal\\\": \\\"Channel\\\",\\n    \\\"PropType\\\": \\\"String\\\"\\n  },\\n  \\\"BackgroundFileName\\\": {\\n    \\\"PropVal\\\": \\\"E:\\\\\\\\Users\\\\\\\\MicroscopeCalibration\\\\\\\\Dark-FF\\\\\\\\MMStack.ome.tif\\\",\\n    \\\"PropType\\\": \\\"String\\\"\\n  },\\n  \\\"Presets\\\": {\\n    \\\"PropVal\\\": [\\n      \\\"Cy5\\\",\\n      \\\"Cherry\\\"\\n    ],\\n    \\\"PropType\\\": \\\"String array\\\"\\n  },\\n  \\\"PresetFiles\\\": {\\n    \\\"PropVal\\\": [\\n      \\\"E:\\\\\\\\Users\\\\\\\\Xiaowei\\\\\\\\FlatfieldImages\\\\\\\\Cy5_20x_From_Cells-B.tif\\\",\\n      \\\"E:\\\\\\\\Users\\\\\\\\Xiaowei\\\\\\\\FlatfieldImages\\\\\\\\Cherry_20x_WideField.tif\\\"\\n    ],\\n    \\\"PropType\\\": \\\"String array\\\"\\n  }\\n}\",\"isEnabled\":true}"],
    "PropType": "String array"
  },
```
which at first glance looks quite similar (and both seem to have excessive escaping).  

Can I simply delete this line?

ChrisWeisiger, 2016-07-28:
Yes, individual lines can (usually? always?) safely be deleted from the profile, as long as you don't accidentally make an invalid JSON file in the process. In your next session the profile entry will be recreated using default values.

The excessive escaping is an unfortunate side-effect of storing JSON-serialized strings inside of strings, as all of the quotation marks in the inner strings must be escaped. These files are not really meant to be read by users, though, so it's not a huge deal.

nicost, 2016-07-28:
OK, that fixed it for one restart.  Next, I quit with the one row active in there.  Upon restart, the pipeline window shows up (as expected), but empty (not expected).  Selecting the Flat-Field Correction plugin causes the Pipeline window to show to rows (both active) with the Flat-Field Correction Processor.  So, the same problem immediately reappears.

Something is not quite right here.

ChrisWeisiger, 2016-07-28:
I'm unable to reproduce this, on Windows or Mac. Could you please share your profile file (the same file you copied that text out of earlier)?

ChrisWeisiger, 2016-07-28:
Thanks for the profile; I was able to reproduce the glitch. In truth, the processor _was_ present, it just wasn't drawn until something else caused the display to update (for example, resizing the pipeline window would cause it to pop into existence).

I have no idea why this happens with your profile but not with mine, but in any case the bug is now fixed.

closed 2016-07-28 by ChrisWeisiger

nicost, 2016-07-28:
Great, thanks!