micro-manager/micro-manager #386: mm2: Acquisition speed dramatically lowered

nicost, 2016-07-29:
We are acquisition data with 2 Andor iXon cameras that are sequentially taking images (i.e. channel 1 has camera 1 as the main cameras, channel 2 has camera 2).  We are running a time-lapse with 100ms exposure each, 100 time points, 0 ms interval.  This takes 58 seconds using beta 3 20160724.  Using beta3 20160728, this takes > 200 seconds.

ChrisWeisiger, 2016-07-29:
I'm unable to reproduce this with demo hardware (not really surprising). Looking at the commit histories for that timespan, the only major changes were the addition of the Alerts system (potentially at fault if the display of how many images have been received thus far is somehow vastly more resource-intensive than expected), and the addition of the Duplicator plugin. There's several smaller commits, but I would not expect any of them to be involved in data acquisition.

You should be able to omit the use of Alerts during MDA by commenting out these lines in MMAcquisition.java's `setProgressText()` method:

         alert_.setText(String.format(
                  "Received " + format + " of %d images",
               imagesReceived_, imagesExpected_));

The alert will still show up at the start of MDA, but it won't be updated, so its resource utilization should not be an issue.

Assuming that doesn't fix things, could you please send us a problem report from each nightly?

nicost, 2016-08-10:
Unable to reproduce this, so this issue can be closed for the moment,

closed 2016-08-11 by marktsuchida