micro-manager/micro-manager #387: mm2: Display hangs when pixel values exceed camera bit depth

marktsuchida, 2016-08-02:
E.g. with a 16-bit image that says it's 12-bit deep but actually contains pixels exceeding 4095.

```
java.lang.ArrayIndexOutOfBoundsException: 15844 in Thread[AWT-EventQueue-0,6,main]
  at org.micromanager.display.internal.ContrastCalculator$InternalCalculator.calculate16BitSingleComponent(ContrastCalculator.java:346)
  at org.micromanager.display.internal.ContrastCalculator$InternalCalculator.calculate(ContrastCalculator.java:177)
  at org.micromanager.display.internal.ContrastCalculator.calculateHistogram(ContrastCalculator.java:485)
  at org.micromanager.display.internal.ContrastCalculator.calculateHistogramWithSettings(ContrastCalculator.java:511)
  at org.micromanager.display.internal.CanvasUpdateQueue.updateHistogram(CanvasUpdateQueue.java:424)
  at org.micromanager.display.internal.CanvasUpdateQueue.scheduleHistogramUpdate(CanvasUpdateQueue.java:391)
  at org.micromanager.display.internal.CanvasUpdateQueue.showImage(CanvasUpdateQueue.java:319)
  at org.micromanager.display.internal.CanvasUpdateQueue.consumeImages(CanvasUpdateQueue.java:246)
  at org.micromanager.display.internal.CanvasUpdateQueue.access$000(CanvasUpdateQueue.java:81)
  at org.micromanager.display.internal.CanvasUpdateQueue$1.run(CanvasUpdateQueue.java:147)
```

hadim, 2016-08-02:
I confirm having a bug like that with an iXon camera. I never reported it because I thought it was because of the computer being very slow (very old Win XP machine).

This bug is very annoying. If you find a fix I can test it to see if it fix my problem.

hadim, 2016-08-02:
I never checked the logs for this bug. I will next time I use it and see if it displays the same errors.

marktsuchida, 2016-08-02:
Thanks, @hadim.

Some more details:
- The bug was originally reported during an acquisition (not an iXon, though), but in my hands it also occurs while scrolling through a reopened image (which is where the above stack trace was taken from).
- In mm2, MDA will correctly save the data itself even if this bug is triggered; the issue is display only.

hadim, 2016-08-02:
Actually I never saw such a bug during MDA but I definitively have it during live acquisition.

At the end maybe it's not related and the slow computer is the culprit here...

PR #388 (2016-08-02): Fix some histogram issues
Fix #387 and some other minor issues.
closed 2016-08-02 by marktsuchida