micro-manager/micro-manager #391: mm2: The functionality of MMFrame should be in the API

marktsuchida, 2016-08-03:
Currently it does this:
- Positioning the window according to its last saved position
- Offsetting said position when there are multiple windows
- Setting up the menu bar (OS X only)
- Saving window position when the window is closed

Plugins should be able to take advantage of this functionality. It should be possible to provide an API that doesn't require subclassing a particular class.

nicost, 2020-04-04:
@marktsuchida can you evaluate if we still want this (I vote yes), and how to go about it if we decide to add it to the API? I'll be happy to help implement and test.

marktsuchida, 2020-04-04:
I remember working on this and the code might still be in master. I had it so you don't have to inherit from `MMFrame`, which is too limiting in the absence of multiple inheritance.

I'll try and see if I can resurrect it.

marktsuchida, 2020-08-28:
See comment in #758 about `WindowPositioning`.