micro-manager/micro-manager #396: Code signing

nicost, 2016-08-10:
When installing Micro-Manager on Windows 10 (and possibly also on other OSs), the user has to go through many scary looking warning signs.  It seems that code signing would prevent those.  There are various options to do so, including ones that are free for open source projects (like https://www.certum.eu/certum/cert,offer_en_open_source_cs.xml)

marktsuchida, 2016-08-11:
Agreed. We should do it for OS X, too, assuming it doesn't cause issues when the `.app` contents change.

henrypinkard, 2022-02-14:
2 tangentially related questions about installers:

1) Do either of you know if it is possible/how difficult it would be to have a command line installer along with the graphical installer?

2) What is the reason that the Windows installer comes with Java binaries but the Mac OS one does not? I'm wondering because headless mode in pycro-manager works quite well on windows because you just point to the correct version of Java in the MM install location, but there's an extra layer of complexity using it on Mac OS

marktsuchida, 2022-02-14:
1. The Windows installer can be run from the command line. Try giving it the option `/?` for details, or see https://jrsoftware.org/ishelp/index.php?topic=setupcmdline. On macOS, you can mount the disk image using `hdiutil`.

2. Probably because traditionally macOS came with Apple's JVM (I'm talking about the early to mid 2000s), and even when that was no longer the case, the OS retained a way to direct the user to install a JVM. I don't think there is any particular reason not to bundle a JRE today.

henrypinkard, 2022-02-14:
> 2\. Probably because traditionally macOS came with Apple's JVM (I'm talking about the early to mid 2000s), and even when that was no longer the case, the OS retained a way to direct the user to install a JVM. I don't think there is any particular reason not to bundle a JRE today.

I think that would be a nice addition for consistency

henrypinkard, 2023-01-11:
Great! I'm happy to test if that would be helpful