micro-manager/micro-manager #425: Album not correctly discarded when "Close All" is used

marktsuchida, 2016-08-19:
To reproduce:
- Snap to Album
- Close All, choose no prompts
- Snap to Album again (nothing appears)

nicost, 2016-08-26:
The underlying cause for this is the implementation that closing a StorageRAM instance is a noop function, and relies on all calling code to delete references to the store.  Since DefaultAlbum uses a static instance with a private member Datastore, DefaultAlbum never lets go of the reference and the store does not close.  
There are many possible solutions, but all I can think of need additions/changes to the API.  For instance, there could be a "IsClosed()" member function of Datastore that will return true once the store is closed.   
This particular issue could also be solved by DefaultDatastore.close() also setting the "isFrozen_" flag. (since DefaultAlbum tests for Datastore.isFrozen().  I don't particularly like that solution, since it would be adding a side effect to the "close()" function.

nicost, 2020-04-04:
Fixed in PR #745

closed 2020-04-04 by nicost