micro-manager/micro-manager #433: Some metadata inserted by 1.4 are lost

nicost, 2016-08-24:
In Micro-Manager 1.4, plugins could insert their own metadata (for instance, the ASI diSPIM plugin inserts metadata concerning the configuration of the system and organization of the data, such as the key-value "NumberOfSides".  In 2.0, those data are inaccessible, basically rendering these datasets unusable.

At the very least, it would be nice if unknown tags would be added to the "UserData" of the SummaryMetadata or the per image Metadata respectively.

marktsuchida, 2016-08-24:
I assume this is for the case of saving a dataset in 1.4 and opening it in 2.0?

nicost, 2016-08-24:
Correct

ChrisWeisiger, 2016-08-24:
What Nico said is *supposed* to happen; there should be code in the MultipageTiffReader that populates the UserData, either with a) whatever keys are indicated by a userDataKeys metadata field (which is created when saving a 2.0 dataset), or b) everything that wasn't a normal metadata field (when loading a dataset that doesn't have the userDataKeys property). Presumably this is malfunctioning somehow.

nicost, 2016-08-24:
It happens for the per-image metadata, not for the summary metadata.  I have a patch that adds that, although likely not as elegant as it was done for the per-image metadata.  I'll see if I can rework it now that I am aware of the per-image code.

nicost, 2016-08-25:
Pull request  #435 addresses this problem.

marktsuchida, 2016-09-09:
It looks like some undesirable items are now getting included in the summary metadata user data. This will need to be fixed.

nicost, 2016-09-09:
What are those undesirable items?  There is now a String array in the code to exclude those undesirable items (and adding the specific ones to them would fix the problem for now), which is not an optimal solution.

marktsuchida, 2016-09-09:
The ones I see so far (not sure if this list is complete) are:
- BitDepth
- CameraTimeout
- ChColors
- Comment
- Depth
- GridColumn
- GridRow
- Height
- IJType
- KeepShutterOpenChannels
- PixelType
- Positions
- ROI
- SlicesFirst
- Source
- TimeFirst
- UUID
- Width

Some of these should be removed entirely (they are already in `SummaryMetadata` or `Metadata`); a few perhaps belong in summary metadata as first-class fields.

nicost, 2016-09-09:
Not adding those to UserData currently would make them disappear (i.e., they will not be listed as metadata at all in 2.0).  At the very least they are now visible again (even if they do not "belong" in the UserData field).  I have no opinion on how this should be organized, but we should get a plan in place for the organization of this data (that deals with managing old datasets nicely).

marktsuchida, 2016-09-09:
My thoughts exactly :)

nicost, 2024-11-12:
Closing as this is much less relevant now.

closed 2024-11-12 by nicost