micro-manager/micro-manager #446: Add NRRD support?

tlnagy, 2016-09-12:
[NRRD (nearly raw raster data)](http://teem.sourceforge.net/nrrd/) is nice file format for preserving metadata that stores the relationships between XYZTC dimensions and their spacing. It looks something like this:

```
NRRD0001
type: double
dimension: 4
sizes: 5 5 5 5
kinds: space space space time
encoding: raw
endian: little
spacings: 1.0 1.0 1.0 nan
space units: "m" "m" "m"
< raw binary data here >
```

It stores that the each pixel in XYZ is 1 meter and that there is no given spacing in the time dimension. It stores the size of each dimension and it is followed by the raw image data following it. The metadata is in simple human-readable ASCII and the raw data can be stored in several different ways (gzipped etc). I know that ImageJ can read/write NRRD files and that the java BioFormats library supports it: https://www.openmicroscopy.org/site/support/bio-formats5.2/formats/nrrd.html

timholy, 2016-09-12:
Perhaps it's worth pointing out that one advantage of NRRD is that it's trivial to `mmap` the data as one giant `XYZTC` array. If I understand OME-TIFF properly (and I may not), this seems less straightforward. Being able to `mmap` the data makes it very easy for many programming languages to access the data in a way that scales to very large data sets.

A possible disadvantage of NRRD is that there may not be that many "reference" example images, particularly for usage of some of the less common options.

marktsuchida, 2016-09-13:
I'm generally a fan of simple formats like this, and agree that the ability to use mmap is very nice. However, there are many more requirements for a format to be usable as a native format in Micro-Manager, at least given the current architecture. For example, it looks like NRRD requires the size of the dataset to be known in advance (at least for all but one axis); this is not the case (among other things, what happens if the user cancels the acquisition?).
(See also: http://lists.openmicroscopy.org.uk/pipermail/ome-devel/2015-May/003293.html )

So it's actually not a trivial task to make the file format "pluggable". That said, I can envision having write-only support for formats other than our native format in the future.

timholy, 2016-09-13:
Excellent considerations. I don't know anything useful about micromanager's architecture, so thanks for filling me in.

With regards to NRRD and the file's size, note that NRRD allows you to have separate header `.nhdr` and `.raw` files; you can always go back and update the `.nhdr` file if acquisition is terminated early. I generally like the separated format better anyway, since that means you can use a text editor to edit the header if necessary without worrying about performance of pushing the data backwards in the file. (I recognize that separated header/data files may not be to all tastes.)

If you want to use a single file, it's still possible as long as you make sure the edited header has the same number of bytes as the original. Premature termination is guaranteed not to require more characters than if the full data set had been acquired, so one can add any necessary padding to make them equal.

tlnagy, 2016-10-10:
@marktsuchida If I started writing a NRRD writer, would it be best to build against the `mm2`-branch?

tlnagy, 2017-09-20:
In case any one is interested, I wrote a parser for Julia that supports OME-TIFF so now you can interface directly with Micromanager image stacks in Julia and have all the metadata loaded as well: https://github.com/tlnagy/OMETIFF.jl

closed 2017-09-20 by tlnagy