micro-manager/micro-manager #450: Problem with numbers formatting

Padoup-Padoup, 2016-09-17:
Hi,

First, many thanks for this piece of software. It is just amazing!

I just compiled micro-manager from source in Debian Wheezy (following [this page of the documentation](https://micro-manager.org/wiki/Linux_installation_from_source_%28Ubuntu%29 ). I am then running micromanager version 1.4. My system is configured with French as the main language (I think believe this is relevant). By default, the comma `,` is used as the decimal delimiter instead of the doc `.`.

The installation went smoothly and I an now using the Wizard to add hardware. When configuring the parameters, some fields are pre-filled with default values (for instance, if I add the `DemoCamera`,the field `MaximumExposureMs` is pre-filled with the value `10000,0000` (note the comma).

Here are the issues:

1. When validating the configuration of a new device in the Wizard, micro-manager will complain about invalid characters entered in the field. It will move to the next step if I manually replace the commas by dots.

2. When saving the configuration, micro-manager will complain again (see screenshot below). To me, it seems that my changes got reverted somewhere after the form validation.

[image]

3. The generated configuration file is invalid for the same reason, decimal separators are saved as commas, and thus micro-manager complains about the wrong number of tokens per line. Note that if I manually correct the configuration file using an text editor, then micro-manager is able to load the file properly.

A (maybe) *easy fix* would be to ignore the system decimal separator and use the dot. This is what people expect when working with scientific software.

Finally, [this issue](https://sourceforge.net/p/micro-manager/mailman/micro-manager-general/thread/49949429.1060206@amodaj.com/) seems related, but is pretty old (2009).

Many thanks!

Padoup-Padoup, 2016-09-29:
To expand a little bit:

- The issues seems to go with any numeric input box (see #455)
- When I run micromanager with the LC_ALL=C locale, the problems goes away, which is an acceptable workaround.

Would there be a compilation flag to set so that the system locale is ignored? I would be happy to test any suggestion to get a more permanent fix.

nicost, 2016-09-29:
This is kind of interesting as Micro-Manager appears to work well in different locals (including the French locale) at least on Windows (and also on Mac OS X last time I tried).  The code converting numbers from the current locale to real numbers and vice versa is in org.micromanager.utils.NumberUtils and it would be interesting to see what you find when you set breakpoints in that code on your system.

Padoup-Padoup, 2016-10-07:
Hmm. Interesting indeed. I will try to have a look at it this and will let you know.

nicost, 2020-02-02:
Closing, but please re-open if this is still an issue.

closed 2020-02-02 by nicost