micro-manager/micro-manager #451: Micro-Manager 2: separate image files specification

sbesson, 2016-09-19:
Hi @marktsuchida @ChrisWeisiger,

Following an [ome-users thread](http://lists.openmicroscopy.org.uk/pipermail/ome-users/2016-September/006169.html) raised by the community, we have been looking into an issue while opening MM 2.0 filesets using Fiji.

The issue we could identify on our side is that the section of the reader which [identifies/groups individual TIFF files](https://github.com/openmicroscopy/bioformats/blob/v5.2.2/components/formats-bsd/src/loci/formats/in/MicromanagerReader.java#L618) relies on the file being named `mg_00000000t_channel_00z.tif`. This naming scheme assumption is made in agreement with the Micro-Manager filesets uploaded to our data repository but also the [Micro-Manager format specification](https://micro-manager.org/wiki/Micro-Manager_File_Formats#Separate_image_files) and the [acquisition code](https://github.com/micro-manager/micro-manager/blob/1.4license/mmstudio/src/org/micromanager/acquisition/TaggedImageStorageDiskDefault.java#L265).

Given the fileset uploaded as part of the thread above but also [another recent fileset](http://downloads.openmicroscopy.org/images/Micro-Manager/1.4.22/thomas/test/test2_sep/Pos0/metadata.txt), it seems that Micro-Manager 2.0 will be departing from this naming scheme assumption and revising the format specification by encoding the series position index into the TIFF filename alongside z-series, channel and timepoint in separate image filesets.

In order to move forward and investigate a fix, there are a few questions on our side:
- is the description above correct and the new expected default in Micro-Manager 2?
- are you planning to allow different variants of this file naming scheme and is there some staging documentation page describing the new specification of the file format?
- is there a corresponding tag/version from which these file layout changes have been implemented allowing backwards compatibility while reading filesets?

Best,
Sebastien

kthorn, 2016-09-20:
Hi Sebastien - 

I uploaded that problematic data set. Based on what you're saying, it sounds like this has to do with whether different XY positions are saved into different folders or not? Under Tools | Options there is a checkbox for "Save XY positions in Separate Image Stack Files". I wonder if the setting of that has an effect on this naming convention?

Kurt

sbesson, 2016-09-22:
Hi @kthorn, having briefly looked at the source code, in addition to the [1.4 naming convention](https://github.com/micro-manager/micro-manager/blob/v2.0.0-beta2/mmstudio/src/main/java/org/micromanager/data/internal/StorageSinglePlaneTiffSeries.java#L347), [new filenames](https://github.com/micro-manager/micro-manager/blob/v2.0.0-beta2/mmstudio/src/main/java/org/micromanager/data/internal/StorageSinglePlaneTiffSeries.java#L325) are generated from a set of coordinates. While the [default coordinates](https://github.com/micro-manager/micro-manager/blob/v2.0.0-beta2/mmstudio/src/main/java/org/micromanager/data/internal/DefaultCoords.java) match your uploaded fileset, the interface seems to be extensible so I would also assume different settings might affect the naming convention.

marktsuchida, 2020-08-31:
The change of file naming for the single-plane TIFF format should have been better documented. For 2.0 we should ensure at least that datasets written by 1.4 open without error. The 1.4 and 2.0 format and naming scheme should also be better documented.

sbesson, 2020-09-07:
Hi Mark, good to hear from you on this issue and hope everything is okay on your side. We will definitely be watching for the upcoming Micro-Manager 2.0 release and the authoritative documentation on the new file format specification.

Our current representation of the state of the intermediate Micro-Manager file formats is still drawn from your email a few years back: http://micro-manager.3463995.n2.nabble.com/A-few-questions-about-Micro-Manager-2-0gamma-td7589403.html#a7589410. I understand Micro-Manager 2.0 will certainly support both versions 1.4 and 2.0. Will either some or all the intermediate file formats mentioned in the thread above also be maintained and/or minimally  readable by Micro-Manager 2.x?

Another related question, Micro-Magellan's latest file format has been mentioned a few times recently notably in the [image.sc](https://forum.image.sc/t/micro-magellan-file-format-compatible-with-omero/33592) forum and I saw the libraries are getting maintained in the codebase. What is the overlap and the relationship between this specification and Micro-Manager 2.0? This might be more a question for @henrypinkard to answer and I am generally happy to migrate it to the https://github.com/micro-manager/NDTiffStorage repository if that's more relevant.

henrypinkard, 2020-09-08:
Probably easiest to explain this from a historical perspective. The format that is now NDTiffStorage was branched off from the Micro-Manager 1.4 multipage TIFF storage about 5 years for the purposes of adding additional features needed for Micro-Magellan--specifically, adding support for multi-resolution pyramids. This was essentially accomplished by creating a different multipage tiff dataset for each resolution level of the pyramid. In the years since, it has been split out into a separate library, which both Micro-Magellan and Pycro-manager use to write data. It has also been modified to better support fast writing of extremely large datasets, and has added support for arbitrary N-dimensional named data axes (i.e. not just channel, z, time, position) in any order, with no knowledge of which axes will be present at the start of the experiment. The cost of this flexibility has been a removal of some features. For example, it no longer writes OMETiffs and doesn't offer the option to split images across tiff files by different Axes. There may be more differences since I'm not familiar with how the Micro-manager multipage tiff format has changed since the split. In addition to the Java classes in NDTiffStorage for reading and writing datasets, there are also Python readers for these datasets described [here](https://pycro-manager.readthedocs.io/en/latest/read_data.html). Given the non-overlapping feature sets NDTiffStorage and Micro-manager multipage tiffs, my guess is that one is not likely to replace the other.

sbesson, 2020-09-15:
Thanks @henrypinkard, the historical background as well as the specifics are very useful to understand the genesis and the motivation of the Micro-Magellan file format.