micro-manager/micro-manager #453: Bug: Setting absolute Z-position in Multi-dimensional acquisition does work

kthorn, 2016-09-20:
I sent a previous problem report about this, but it is easily reproducible with the demo configuration. Open Multi-dimensional acquisition and stage control. Check Z-stacks and select absolute position in Multi-dimensional acquisition.  Use the stage control dialog to move the Z-stage to a non-zero position. Click set in Z-stacks: set position is always 0 regardless of the actual Z-stage position. 

Based on testing on multiple nightly builds, I believe this bug was introduced on 8/15.

nicost, 2016-09-29:
Indeed, this bug was caused by re-organization of the startup sequence in the Micro-Manager code.  As a consequence, the function MMStudio.initializeGUI() is called before the object staticInfo_ is created.  Since the current z stage in the acquisition engine is set in initializeGUI (which fails, because there is no staticInfo_ and at that point the engine_ is not created yet), the acquisition engine is not aware what the current z stage is.  

There are many ways to fix this, and I am not sure what the best approach is (this is more of a design decision for Mark) .  However, I submitted a pull request (#456), in which I work around the problem by specifically setting the zDrive at the moment the engine is created.

nicost, 2016-12-04:
Since the PR was merged, I'll close this ticket.

closed 2016-12-04 by nicost