micro-manager/micro-manager #455: `Arduino-DAC1-Volts`  only take integer values

Padoup-Padoup, 2016-09-27:
Hello,

I am using an Arduino to control a z-stage. Here is the issue I encounter:

When I try to command the Arduino DAC directly with the cursor that goes from 0 to 5 (volts), the cursor will only settle on integer values. Similarly, the DAC output will only settle on the corresponding voltages.

Similarly, when I use the DA Z stage adapter, the cursor only settles to integer values (in this case, this is less of a problem since there are 200 integers instead of 6 within the accepted range).

I'd be happy to provide any additional information.

I am using this version of micromanager:
> MM Studio version: 1.4.x dev
> MMCore version 8.3.0
> Device API version 67, Module API version 10
> (compiled from source, GNU/Linux machine)

nicost, 2016-09-29:
That sounds like a locale issue.  What language is your system set to?  Can you set the voltage to the correct value using something like:

mmc.setProperty("ArduinoDA", "Volts", "1.75"); 

B.t.w., did you use the TLV5618 chip to get a DA?

Padoup-Padoup, 2016-09-29:
Hi Nico,

I didn't think about directly setting the property. And it works. Thank you!
My system is in French (where the decimal separator is a comma **,** and not a **.** ).
By the way:

1. I had to run `mmc.setProperty("Arduino-DAC1", "Volts", "1.75"); ` on my system (note the different property name).
2. When I run micromanager with the `LC_ALL=C` locale, the problem goes away.
3. This seems to relate with micromanager not properly handling locales: #450 

So far I've just edited the `.ino` file to use the PWM together with a RC circuit for smotthing, but depending on the results we might switch to the TLV5618.

Thanks again!

closed 2016-09-29 by Padoup-Padoup