micro-manager/micro-manager #458: MDA hang

nicost, 2016-09-29:
Regularly, running an MDA results in what appears to be a deadlock involving the EDT (i.e., Miro-Manager needs to be killed).  This only happens when using actual hardware and seems quite reproducible (until I tried to reproduce it;().

BioinfoTongLI, 2016-10-20:
Hi, nicost,

First, thanks all the contributors for this super cool tools of microscopy.
I really enjoy using.

I confirm this bug. But with demo configuration also, not necessarily with actual hardware.
Sometimes rebooting MM solve the problem 
Sorry i didn' t look into detail the code. But it seems do a snap before live mode or MDA helps to reduce the possibility of hanging. 
Oh, yes. Live mode also hangs (tested on Mac, linux and windows)
I'm using the latest daily build. 
Develop environment : IntelliJ
Here is the error i got.

> [IFO,App] EDTHangLogger: Event handling has exceeded at least 1003 ms (currently 5504 ms since heartbeat timebase)
>                                               [       ] Stack traces follow (note: thread states queried later than stack traces)
>                                               [       ] Thread 2 [Reference Handler] WAITING
>                                               [       ]   at java.lang.Object.wait(Native Method)
>                                               [       ]   at java.lang.Object.wait(Object.java:503)
>                                               [       ]   at java.lang.ref.Reference$ReferenceHandler.run(Reference.java:133)
>                                               [       ] Thread 3 [Finalizer] WAITING
>                                               [       ]   at java.lang.Object.wait(Native Method)
>                                               [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:135)
>                                               [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:151)
>                                               [       ]   at java.lang.ref.Finalizer$FinalizerThread.run(Finalizer.java:209)
>                                               [       ] Thread 4 [Signal Dispatcher] RUNNABLE
>                                               [       ] Thread 12 [RMI TCP Accept-0] RUNNABLE
>                                               [       ]   at java.net.PlainSocketImpl.socketAccept(Native Method)
>                                               [       ]   at java.net.AbstractPlainSocketImpl.accept(AbstractPlainSocketImpl.java:398)
>                                               [       ]   at java.net.ServerSocket.implAccept(ServerSocket.java:530)
>                                               [       ]   at java.net.ServerSocket.accept(ServerSocket.java:498)
>                                               [       ]   at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.executeAcceptLoop(TCPTransport.java:399)
>                                               [       ]   at sun.rmi.transport.tcp.TCPTransport$AcceptLoop.run(TCPTransport.java:371)
>                                               [       ]   at java.lang.Thread.run(Thread.java:745)
>                                               [       ] Thread 13 [RMI Reaper] WAITING
>                                               [       ]   at java.lang.Object.wait(Native Method)
>                                               [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:135)
>                                               [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:151)
>                                               [       ]   at sun.rmi.transport.ObjectTable$Reaper.run(ObjectTable.java:351)
>                                               [       ]   at java.lang.Thread.run(Thread.java:745)
>                                               [       ] Thread 14 [GC Daemon] TIMED_WAITING
>                                               [       ]   at java.lang.Object.wait(Native Method)
>                                               [       ]   at sun.misc.GC$Daemon.run(GC.java:117)
>                                               [       ] Thread 15 [Java2D Disposer] WAITING
>                                               [       ]   at java.lang.Object.wait(Native Method)
>                                               [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:135)
>                                               [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:151)
>                                               [       ]   at sun.java2d.Disposer.run(Disposer.java:145)
>                                               [       ]   at java.lang.Thread.run(Thread.java:745)
>                                               [       ] Thread 17 [AWT-XAWT] RUNNABLE
>                                               [       ]   at sun.awt.X11.XToolkit.waitForEvents(Native Method)
>                                               [       ]   at sun.awt.X11.XToolkit.run(XToolkit.java:541)
>                                               [       ]   at sun.awt.X11.XToolkit.run(XToolkit.java:505)
>                                               [       ]   at java.lang.Thread.run(Thread.java:745)
>                                               [       ] Thread 18 [AWT-EventQueue-0] BLOCKED
>                                               [       ]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplayedImageTo(DefaultDisplayWindow.java:698)
>                                               [       ]   at org.micromanager.display.internal.DefaultDisplayWindow.requestRedraw(DefaultDisplayWindow.java:710)
[... cut ...]
>                                               [       ] Thread 44 [FPS display blank] WAITING
>                                               [       ]   at java.lang.Object.wait(Native Method)
>                                               [       ]   at java.lang.Object.wait(Object.java:503)
>                                               [       ]   at java.util.TimerThread.mainLoop(Timer.java:526)
>                                               [       ]   at java.util.TimerThread.run(Timer.java:505)

nicost, 2016-10-21:
You can apply pull request #459.  Removing that lock should avoid the deadlock that you (and I and likely many others) ran into.  I have not yet discovered downsides, but the threads and locks in the display code needs careful study.

nicost, 2016-12-04:
I think this has been fixed now.  Please do open a new ticket whenever you experience a MDA hang with the current code

closed 2016-12-04 by nicost