micro-manager/micro-manager #461: Updated drivers for Thorlabs positioners/ stages

derinsevenler, 2016-10-01:
Thorlabs has recently (like summer 2016) made changes to some of it's hardware products and motion control software for motorized positioning. They have superseded their 'APT' control software with a new API called '[Kinesis](https://www.thorlabs.com/software_pages/ViewSoftwarePage.cfm?Code=Motion_Control)'.

On the hardware side, they have also replaced their 'T-cube' line of motion controllers with a new line called '[K-cube](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=2419)'. Although Thorlabs claims that the K-cube controllers are still compatible with the old APT software, the existing micro-manager plugin 'ThorlabsAPTStage' doesn't seem to work with at least some models. My observation is basically identical to the one described on [this recent forum discussion](https://sourceforge.net/p/micro-manager/mailman/message/34941492/).

I recently made a branch and uploaded a new device adapter that uses the new Kinesis software instead of APT to talk with K-cube controllers, it's [located here](https://github.com/derinsevenler/micro-manager/tree/ThorlabsKinesisStage). 

Right now the adapter will only work with some particular stages ([MTS25 & MTS50](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=3002)), so it's not a general solution for everyone using Thorlabs positioners. But, this might be helpful for some people as-is, and might have some value as a starting point for a more general Kinesis adapter. 

I can write up some documentation for it, if there is interest. Also, I'm wondering if anyone on Micro-Manager's core team has ever reached out to Thorlabs about getting help writing these drivers?

nicost, 2020-01-31:
It looks like Mark is working on a Kinesis device adapter.

nicost, 2020-04-04:
@marktsuchida can you evaluate the situation with the Kinesis?  There still does not appear to be a Kinesis or ThorlabsKinesis adapter in the repository, which is a pity.

marktsuchida, 2020-04-04:
Yes, I have a Kinesis device adapter that should work with most of the motor devices, but it requires Visual Studio 2015+ because getting around the poor design of the Thorlabs SDK (which looks like the product of massive copy-and-paste programming) would be impractical without C++11 (at least for me).

I haven't gotten around to making this public because it would take significant work to integrate it into the nightly build at the moment. For now I can put the source code in a separate repo and at least provide a downloadable binary -- not a good trend to start haphazardly, but perhaps better than not having anything?

nicost, 2020-04-04:
Great plan!  Would be great if you can make a page on the micro-manager.org site and provide a link and instructions there.  Hopefully, we'll move to VS>=2015 soon!

jorishey1234, 2020-05-06:
> For now I can put the source code in a separate repo and at least provide a downloadable binary

It would be great to share it ! I have a few Thorlabs displacement tables (LTS300 and MTS50 with KDC101) and I wish to control in micromanager...

yfukai, 2021-02-03:
Hi, we have Thorlabs KST101 controller and trying to control it from MM. The idea of having device adapter sounds great. @marktsuchida, would you mind sharing the code for the Kinesis driver? We would like to see how it works with our Visual Studio 2019 and KST101.

ckolluru, 2021-03-01:
Hi,

I'm quite new to micro-manager. I'd like to control a MTS-25 stage with KDC101 controller from micro-manager. I've installed Micromanager v2.0.0-gamma1. 

I've taken a look at Derin's repo, and found the folder with the Visual Studio project (ThorlabsKinesisStage), but I'm unsure how I can integrate it into MicroManager. Can someone please give me some instructions?

Thanks,
Chaitanya

amanninen09, 2021-05-25:
Hi  there, is there an update regarding a Kinesis driver for KST101 @marktsuchida ?

marktsuchida, 2021-05-26:
@amanninen09 Thanks you for reminding me. I'm looking at the code now and it looks like it no longer works with the latest Kinesis SDK, so will see if I can fix it.

marktsuchida, 2021-05-28:
This is now available at https://github.com/micro-manager/mmdev-ThorlabsKinesis.
Please see the readme file there for details, and report any issues in that repository.
When it becomes technically feasible to do so, we will include it in the normal Micro-Manager download.

@nanthony21 @nicost I kind of started the convention of using the prefix `mmdev-` plus the device adapter name (as displayed in the HCW) for repositories containing a single device adapter. I'm open to other prefixes but let's keep it consistent, should any other adapters require this treatment.

amanninen09, 2021-06-04:
I moved the mmgr_dal_ThorlabsKinesis.dll file from the Releases page into the Micro-Manger-2.0gamma folder but when I checked the CoreLogs folder in the Micro-Manger folder I saw this error: 
Unable to load ThorlabsKinesis library: Failed to load device adapter "ThorlabsKinesis" [ Failed to load module "C:\Program Files\Micro-Manager-2.0gamma\mmgr_dal_ThorlabsKinesis.dll" [ The module, or a module it depends upon, could not be found (Windows error: The specified module could not be found.) ] ] 
Perhaps there is another dependent module missing?

marktsuchida, 2021-06-04:
@amanninen09 Thanks for the log details! mmgr_dal_ThorlabsKinesis.dll does not directly (statically) link to the Thorlabs DLLs, so this error message at this stage might mean that you need to install (or update) the Microsoft Visual C++ Redistributable ([64-bit version](https://aka.ms/vs/16/release/vc_redist.x64.exe) / [info](https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0)). Could you try that? I'll update the instructions if it works.

amanninen09, 2021-06-04:
Thanks Mark that did the trick I was able to add the device in the HC Wizard and set the KST101 as the default focus stage in step 3. I haven't been able to actually move the stage through MM yet do you have advise on the user interface for that? I tried it on the Kinesis app and it still works there, and fyi afterward it was buggy getting the KST101 to re-connect with MM - had to close both of the apps to reset but it did reconnect then.

marktsuchida, 2021-06-04:
Yeah, once an app connects to the stage, it probably has to be quit before other apps can cleanly connect.
Great to hear it worked. The user interface for moving the stage is called Stage Control, which you can access either from the Tools menu or the four-way-arrow icon in the main window, under "Stage".

closed 2021-06-04 by marktsuchida