micro-manager/micro-manager #469: Missing ChannelGroupChangedEvent

nicost, 2016-10-20:
It would be very useful to be able to subscribe to an event that is fired whenever the core channel group is changed.

nicost, 2017-03-04:
Ran into this again.  There is an internal event Channel Group event with the following comment:  "This event is posted by AcquisitionWrapperEngine when its setChannelGroup method is called. We don't expose it in the API to avoid giving the impression that it is posted whenever the channel group changes for any reason, and because the API also doesn't provide access to that setChannelGroup method."
It seems straight forward to add a callback in the core and wire that up to the Java code and acquisition engine. Am I missing something?

marktsuchida, 2020-04-05:
I think the problem is that "channel group" is an acquisition dialog and main window concept, not part of the Core. It used to be only an acquisition concept, but then it had to be added to the main window (in early 2.0beta) to make the until-then buggy "sync exposure" feature to actually behave logically.

I'm not totally opposed to moving it into the Core, but adding the callback should be possible without putting it into the Core, and I think a better approach would be to first add the event and make sure every piece of code that handles channel groups correctly uses it.

nicost, 2020-04-10:
Actually, it is part of the core.  See https://valelab4.ucsf.edu/~MM/doc/MMCore/html/class_c_m_m_core.html:

> std::string getChannelGroup ()
> void setChannelGroup (const char *channelGroup)  throw (CMMError)

I'll create the callback from the Core that results in posting the OnChannelGroupChangedEvent on the event bus.  I am tempted to remove the internal ChannelGroupEvent, since it complicates things.  Only tricky part is that the UI element that changes the ChannelGroup should not respond to the OnChannelGroupChangeEvent that it causes.

nicost, 2020-04-10:
Event added in PR #770.  Works as advertised (used by Mainframe, MDA, and RatioImaging plugin).

closed 2020-04-10 by nicost