micro-manager/micro-manager #475: setAcquisitionSettings(settings) ignores settings.channelGroup

kthorn, 2016-11-03:
Here is an example using the demo configuration with MM2.0:
bsh % acq = mm.acquisitions();
<org.micromanager.acquisition.internal.DefaultAcquisitionManager@26a30589>

### Get the current Settings
bsh % settings = acq.getAcquisitionSettings();
<org.micromanager.acquisition.SequenceSettings@77e3cabd>

### Channel Group and Interval Settings are..
bsh % settings.channelGroup;
<Channel>
bsh % settings.intervalMs;
<0.0>

### Set to new values..
bsh % settings.channelGroup = "Objective";
<Objective>
bsh % settings.intervalMs = 10;
<10>
### Apply new settings
bsh % acq.setAcquisitionSettings(settings);

### Check whether that have been applied.  Interval has, channelGroup has not been applied.
bsh % settings_new = acq.getAcquisitionSettings();
<org.micromanager.acquisition.SequenceSettings@454322ba>
bsh % settings_new.intervalMs;
<10.0>
bsh % settings_new.channelGroup;
<Channel>

The issue seems to be that getSequenceSettings in AcquisitionWrapperEngine.java reads the channelGroup from the core, but setSequenceSettings does not set it in the core.

kthorn, 2016-11-03:
Fixed in PR #479. In fixing this I noticed a number of other inconsistent behaviors. First, channelGroup is not checked to see if it matches the channels, so by manually changing the sequence settings it is possible to set channels that do not belong to the channelGroup. Second, calling setSequenceSettings with no channels sets useChannels_ to false, but does not update the checkbox on the UI.

nicost, 2016-12-04:
Since the PR is merged, I'll close this ticket.  If necessary, please do open another ticket with remaining issues.

closed 2016-12-04 by nicost