micro-manager/micro-manager #480: Deadlock while starting an acquisition

nicost, 2016-11-04:
I have a potential fix, needs testing:

debug:
2016-11-03T17:23:11.738566 tid5020 [IFO,LogManager] Enabled logging to stderr
Asked to load file at path C:\Program Files\Micro-Manager-2.0beta\GlobalUserProfile.txt which doesn't exist; instead providing a default empty PropertyMap.
2016-11-03T17:23:11.832167 tid5020 [IFO,LogManager] Enabled primary log file C:\Program Files\Micro-Manager-2.0beta\CoreLogs\CoreLog20161103T172311_pid4248.txt
2016-11-03T17:23:11.832167 tid5020 [IFO,App] User: Vale Lab
2016-11-03T17:23:11.832167 tid5020 [IFO,App] Host: Uberscope
2016-11-03T17:23:11.832167 tid5020 [IFO,App] MM Studio version: 2.0.x dev
---- snip ---
2016-11-03T17:24:25.703410 tid2396 [IFO,App] EDTHangLogger: Missed heartbeat; waiting to see if we are stuck on a single event
2016-11-03T17:24:26.704467 tid2396 [IFO,App] EDTHangLogger: Event handling has exceeded at least 1000 ms (currently 5500 ms since heartbeat timebase)
                                   [       ] Stack traces follow (note: thread states queried later than stack traces)
                                   [       ] Thread 2 [Reference Handler] WAITING
                                   [       ]   at java.lang.Object.wait(Native Method)
                                   [       ]   at java.lang.Object.wait(Object.java:485)
                                   [       ]   at java.lang.ref.Reference$ReferenceHandler.run(Reference.java:116)
                                   [       ] Thread 3 [Finalizer] WAITING
                                   [       ]   at java.lang.Object.wait(Native Method)
                                   [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:118)
                                   [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:134)
                                   [       ]   at java.lang.ref.Finalizer$FinalizerThread.run(Finalizer.java:171)
                                   [       ] Thread 4 [Signal Dispatcher] RUNNABLE
                                   [       ] Thread 5 [Attach Listener] RUNNABLE
                                   [       ] Thread 13 [Java2D Disposer] WAITING
                                   [       ]   at java.lang.Object.wait(Native Method)
                                   [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:118)
                                   [       ]   at java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:134)
                                   [       ]   at sun.java2d.Disposer.run(Disposer.java:127)
                                   [       ]   at java.lang.Thread.run(Thread.java:662)
                                   [       ] Thread 14 [AWT-Shutdown] WAITING
                                   [       ]   at java.lang.Object.wait(Native Method)
                                   [       ]   at java.lang.Object.wait(Object.java:485)
                                   [       ]   at sun.awt.AWTAutoShutdown.run(AWTAutoShutdown.java:265)
                                   [       ]   at java.lang.Thread.run(Thread.java:662)
                                   [       ] Thread 15 [AWT-Windows] RUNNABLE
                                   [       ]   at sun.awt.windows.WToolkit.eventLoop(Native Method)
                                   [       ]   at sun.awt.windows.WToolkit.run(WToolkit.java:293)
                                   [       ]   at java.lang.Thread.run(Thread.java:662)
                                   [       ] Thread 18 [AWT-EventQueue-0] BLOCKED
                                   [       ]   at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
                                   [       ]   at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
                                   [       ]   at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
                                   [       ]   at com.google.common.eventbus.EventBus.post(EventBus.java:275)
                                   [       ]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplaySettings(DefaultDisplayWindow.java:791)
                                   [       ]   at org.micromanager.display.internal.DefaultDisplayWindow.setDisplaySettings(DefaultDisplayWindow.java:773)
                                   [       ]   at org.micromanager.display.internal.LUTMaster.updateDisplayLUTForChannel(LUTMaster.java:319)
                                   [       ]   at org.micromanager.display.internal.LUTMaster.updateDisplayLUTs(LUTMaster.java:289)
                                   [       ]   at org.micromanager.display.internal.LUTMaster.initializeDisplay(LUTMaster.java:261)
                                   [       ]   at org.micromanager.display.internal.CanvasUpdateQueue.showImage(CanvasUpdateQueue.java:333)
                                   [       ]   at org.micromanager.display.internal.CanvasUpdateQueue.consumeImages(CanvasUpdateQueue.java:219)
                                   [       ]   at org.micromanager.display.internal.CanvasUpdateQueue.access$000(CanvasUpdateQueue.java:71)
                                   [       ]   at org.micromanager.display.internal.CanvasUpdateQueue$1.run(CanvasUpdateQueue.java:158)
                                   [       ]   at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:209)
                                   [       ]   at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:672)
                                   [       ]   at java.awt.EventQueue.access$400(EventQueue.java:81)
                                   [       ]   at java.awt.EventQueue$2.run(EventQueue.java:633)
                                   [       ]   at java.awt.EventQueue$2.run(EventQueue.java:631)
                                   [       ]   at java.security.AccessController.doPrivileged(Native Method)
                                   [       ]   at java.security.AccessControlContext$1.doIntersectionPrivilege(AccessControlContext.java:87)
                                   [       ]   at java.awt.EventQueue.dispatchEvent(EventQueue.java:642)
                                   [       ]   at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:269)
                                   [       ]   at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:184)
                                   [       ]   at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:174)
                                   [       ]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:169)
                                   [       ]   at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:161)
                                   [       ]   at java.awt.EventDispatchThread.run(EventDispatchThread.java:122)
                                   [       ] Thread 21 [org.netbeans.modules.debugger.jpda.visual AWT Access Loop] TIMED_WAITING
                                   [       ]   at java.lang.Thread.sleep(Native Method)
                                   [       ]   at org.netbeans.modules.debugger.jpda.visual.remote.RemoteAWTService$AWTAccessLoop.run(RemoteAWTService.java:236)
[... 99 lines cut ...]
                                   [       ]   at com.google.common.eventbus.EventSubscriber.handleEvent(EventSubscriber.java:74)
                                   [       ]   at com.google.common.eventbus.SynchronizedEventSubscriber.handleEvent(SynchronizedEventSubscriber.java:47)
                                   [       ]   at com.google.common.eventbus.EventBus.dispatch(EventBus.java:322)
                                   [       ]   at com.google.common.eventbus.EventBus.dispatchQueuedEvents(EventBus.java:304)
                                   [       ]   at com.google.common.eventbus.EventBus.post(EventBus.java:275)
                                   [       ]   at org.micromanager.internal.utils.PrioritizedEventBus.post(PrioritizedEventBus.java:63)
                                   [       ]   at org.micromanager.data.internal.DefaultDatastore.putImage(DefaultDatastore.java:202)
                                   [       ]   at org.micromanager.data.internal.pipeline.DefaultPipeline.insertImage(DefaultPipeline.java:111)
[... cut ...]
                                   [       ]   at java.lang.Thread.run(Thread.java:662)
                                   [       ] Thread 106 [FPS display blank] WAITING
                                   [       ]   at java.lang.Object.wait(Native Method)
                                   [       ]   at java.lang.Object.wait(Object.java:485)
                                   [       ]   at java.util.TimerThread.mainLoop(Timer.java:483)
                                   [       ]   at java.util.TimerThread.run(Timer.java:462)

kthorn, 2016-11-04:
I see what looks like the same hang in the latest nightly build. I was able to acquire a few images before it occurred, so it appears to be sporadic.

nicost, 2016-11-04:
It is useful to look at the threads that are "BLOCKED" (hopefully recorded in the Corelog).  If the EDT (and a second thread) are blocked at the same place as in the trace above, then it is the smae deadlock (there may be others, and fixes to this deadlock may uncover yet other deadlocks).  B.t.w., this logging is very nice!

nicost, 2016-11-04:
PR #482 hasa fix for this deadlock.  On this particular microscope system it now appears 100% of the time.  After applying the PR I have not seen it anymore (or any other deadlock, however, this is only with limited testing).

nicost, 2016-12-04:
Merged #482

closed 2016-12-04 by nicost