micro-manager/micro-manager #485: mm2 : MDA with multiple XY Position into separates files

hadim, 2016-12-04:
I would like to run MDA with multiple XY Position and be able to save separately the dataset into a different files for each XY Position.

There is a lot of reasons behind that but one of them is that different XY Position are totally uncorrelated and are analysed independently from each other in my case.

In my opinion this should be a core option of mm2 but I'll understand if you don't this as an internal mm2 feature.

This is why I propose you to write a `ProjectorPlugin` that would split a `DataStore` into multiples `DataStore` : one for each XY Position.

Would you be interested in such plugin for mm2 ?

@nicost any opinions on this ?

nicost, 2016-12-04:
There is an option (Tools > Options) that lets you save XY positions in individual TIFF files.  Did you miss that or do you want something else?

hadim, 2016-12-04:
I know this option but the issue is that the files generated are not independent from each other because only the first one contains the OME Metadata and so Fiji only open the first one with the correct CZT dimensions. The others files are detected 3D only stack of images.

What I would like is a different `Datastore` for each positions.

hadim, 2016-12-04:
See also https://github.com/micro-manager/micro-manager/pull/489 for a proof of concept.

marktsuchida, 2020-08-31:
It's best to avoid adding special-case features. If there is going to be an option to split the dataset by position, it should work cleanly with any axis or axes. But I feel that this is specialized enough that it should be handled by scripting, perhaps by offering an easier way to implement a proxy `Datastore` (backed by a `Datastore` per position) -- I vaguely remember that method being discussed elsewhere. Closing for now.

closed 2020-08-31 by marktsuchida