micro-manager/micro-manager #498: Rules for using Core notifications without deadlock are unclear

marktsuchida, 2016-12-14:
In mm2, we might consider using a second, asynchronous event bus (AsyncEventBus) for these to avoid deadlock.

nicost, 2020-04-04:
@marktsuchida Shall we have a look at this now?

marktsuchida, 2020-04-04:
Yes. I think there are mainly two options: an async event bus, or `SwingUtilities.invokeLater()`, i.e., always run the callbacks on the EDT. For GUI updates the latter makes more sense, but if most callbacks need to do further background processing/querying before updating the GUI anyway, perhaps it is better to use an async event bus.

nicost, 2020-04-04:
It looks like all callbacks in org.micromanager.events.internal.CoreEventCallback are on the EDT.

Close ticket?

marktsuchida, 2020-04-04:
Actually it's probably best to update that class so that it uses `invokeLater` unconditionally, even when already on the EDT. Otherwise there could be cases where a callback on the EDT calls back into the Core, resulting in further callbacks, which seems dangerous.

nicost, 2020-04-05:
Implemented in PR #754

closed 2020-04-05 by nicost