micro-manager/micro-manager #505: Trouble building mm2 on openSUSE

tlnagy, 2016-12-21:
I would like to setup Micro-Manager 2 on a fresh install of openSUSE Leap 42.2 installation. I cloned the repo and checked out the mm2 branch. Installed the following packages:

```
sudo zypper in gcc-c++ gcc make boost-devel libz1 automake libtool swig ant python-devel python-numpy-devel libusb-compat-devel libgphoto2-devel libdc1394-devel libhidapi-devel
```

Running `./autogen.sh`

```
Bootstrapping autoconf/automake build system for Micro-Manager...
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: adding subdirectory DeviceAdapters to autoreconf
autoreconf: Entering directory `DeviceAdapters'
autoreconf: running: aclocal --force -I ../m4
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `../m4'.
libtoolize: copying file `../m4/libtool.m4'
libtoolize: copying file `../m4/ltoptions.m4'
libtoolize: copying file `../m4/ltsugar.m4'
libtoolize: copying file `../m4/ltversion.m4'
libtoolize: copying file `../m4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: automake --add-missing --copy --force-missing
autoreconf: Leaving directory `DeviceAdapters'
autoreconf: configure.ac: adding subdirectory SecretDeviceAdapters to autoreconf
autoreconf: Entering directory `SecretDeviceAdapters'
autoreconf: configure.ac: not using Autoconf
autoreconf: Leaving directory `SecretDeviceAdapters'
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
autoreconf: Leaving directory `.'
Bootstrapping complete; now you can run ./configure
If you would like to install Micro-Manager as an ImageJ plugin, run
  ./configure --enable-imagej-plugin=/path/to/ImageJ
```

Followed by `./configure`

```
checking build system type... x86_64-suse-linux-gnu
checking host system type... x86_64-suse-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
[... 473 lines cut ...]
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/oweinerlab/mm/micro-manager/MMDevice/unittest'
make[1]: Leaving directory '/home/oweinerlab/mm/micro-manager/MMDevice'
Making all in MMCore
make[1]: Entering directory '/home/oweinerlab/mm/micro-manager/MMCore'
Making all in .
make[2]: Entering directory '/home/oweinerlab/mm/micro-manager/MMCore'
make[2]: Nothing to be done for 'all-am'.
make[2]: Leaving directory '/home/oweinerlab/mm/micro-manager/MMCore'
make[1]: Leaving directory '/home/oweinerlab/mm/micro-manager/MMCore'
Making all in MMCorePy_wrap
make[1]: Entering directory '/home/oweinerlab/mm/micro-manager/MMCorePy_wrap'
/bin/sh ../libtool  --tag=CXX   --mode=link g++  -g -O2 -module -avoid-version -shrext .so  -o _MMCorePy.la -rpath /usr/local/lib/micro-manager _MMCorePy_la-MMCorePy_wrap.lo ../MMCore/libMMCore.la -ldl 
libtool: link: g++  -fPIC -DPIC -shared -nostdlib /usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../lib64/crti.o /usr/lib64/gcc/x86_64-suse-linux/4.8/crtbeginS.o  .libs/_MMCorePy_la-MMCorePy_wrap.o  -Wl,--whole-archive ../MMCore/.libs/libMMCore.a -Wl,--no-whole-archive  -L/usr/lib64 -lboost_system -lboost_date_time-mt -lboost_thread -ldl -L/usr/lib64/gcc/x86_64-suse-linux/4.8 -L/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/lib -L/usr/lib64/gcc/x86_64-suse-linux/4.8/../../.. -lstdc++ -lm -lc -lgcc_s /usr/lib64/gcc/x86_64-suse-linux/4.8/crtendS.o /usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../lib64/crtn.o  -O2   -Wl,-soname -Wl,_MMCorePy.so -o .libs/_MMCorePy.so
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: ../MMCore/.libs/libMMCore.a(Debayer.o): relocation R_X86_64_32 against `.rodata' can not be used when making a shared object; recompile with -fPIC
../MMCore/.libs/libMMCore.a(Debayer.o): error adding symbols: Bad value
collect2: error: ld returned 1 exit status
Makefile:474: recipe for target '_MMCorePy.la' failed
make[1]: *** [_MMCorePy.la] Error 1
make[1]: Leaving directory '/home/oweinerlab/mm/micro-manager/MMCorePy_wrap'
Makefile:509: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
```

I wasn't able to decipher the error here. Any suggestions would be appreciated!

marktsuchida, 2016-12-21:
Not sure but this might be relevant: http://unix.stackexchange.com/questions/242755/error-while-building-snipersim-relocation-r-x86-64-32s-against-rodata-str1-1

csachs, 2017-02-17:
Hi, does this problem still persist?
I just tried to reproduce it by installing openSUSE Leap 42.2 in a VM, however, for me, mm2 built flawlessly (with the packages installed @tlnagy described).
Btw, for me it autodetected java (installed as a dependency for ant, I assume)

tlnagy, 2017-02-17:
I haven't tried since then so I'm glad to hear that you couldn't reproduce. Was your installation on a fresh VM with only the packages listed above or did you have other things installed?

csachs, 2017-02-18:
Yes, I purposely installed it fresh just to test this, and just the packages (+ git) you have mentioned.

tlnagy, 2017-02-18:
I'm glad to hear. I haven't had to time to check this out, so I'm going to go ahead and close this for now. I'll reopen it if I run into problems. Thanks for letting me know @csachs

closed 2017-02-18 by tlnagy