micro-manager/micro-manager #506: Locale is (partially) ignored in Stage Control Frame

nicost, 2016-12-21:
The conversion between text and numbers used to be delegated to the NumberUtils class (in Utilities).  This class takes the runtime locale into account.  The Stage Control plugin in 1.4 had its own implementation (likely to avoid the use of non-api functions).  During the port to 2.0 this code was changed and now uses the locale to convert numbers to text, but ignores the locale when converting text to numbers.  This obviously leads to problems with certain locales.  An simple solution is to use the NumberUtils class again, but I would like to know if that is the agreed upon strategy in 2.0 or if there is another plan.

marktsuchida, 2016-12-21:
Yes, `NumberUtils` should be used. We might eventually move it to provide these conversions in the API, but all code using it will be updated at that time.
It looks like the use of `NumberFormats` was removed in this commit: 9633651f0338b73511c9ece2d90b6551215e4511 (but that was a local (fake?) copy of `NumberUtils`).

nicost, 2016-12-22:
Sorry, I did not intend to start a blame war, but the history is that the Stage Control Plugin in 1.4 used its own version of NumberUtils (since it was not in the api).  So little was used, that all the relevant functionality was in-lined in the commit you mentioned.  However, in the port to 2.0, the use of NumberFormat was abandoned and text was converted into a double using a bare Double.parseDouble() (which we know leads to trouble with non-US locales).  So currently, the code works correctly in 1.4 but incorrectly in 2.0.  I am all for consolidating everything to NumberUtils and happy to work on a PR.

marktsuchida, 2016-12-22:
Thanks, Nico. I somehow thought the removal of `NumberUtils` had something to do with the UI changes in that commit (which predates the 2.0 branch) -- didn't mean to assign blame. In any case I don't think there is any reason not to use `org.micromanager.internal.utils.NumberUtils`, especially given that StageControl now lives inside MMStudio.

closed 2020-01-31 by nicost