micro-manager/micro-manager #509: Shutter does not open when live is first started

kthorn, 2017-01-03:
When first starting live mode after initially launching Micro-manager, the selected shutter does not open. Stopping and restarting live mode results in the shutter opening properly. This is tested in the 1/2/2017 build of MM2.0.

nicost, 2017-01-03:
I vividly remember looking at this bug and even concluding what caused it (I thought it was written down in one of the issues or PRs).  I think it had something to do with the initialization sequence.  Petty that my memory is not better....

marktsuchida, 2017-01-04:
@kthorn What device is this shutter(s)? I'm wondering if it is device-specific.

kthorn, 2017-01-05:
@marktsuchida This was on our spinning disk confocal using Nico's NNLC board for control.  I can check on some of our other systems to see if its device specific or not.  That might take me until early next week.

closed 2020-01-31 by nicost