micro-manager/micro-manager #516: Bad preprocessor in IIDC device adapters causing mm2 build to fail in Linux

kmdouglass, 2017-01-26:
# Problem description

Building mm2 for Linux x64 will outright fail due to a supposedly bad preprocessor directive in the file **MMIIDCCamera.cpp**. The offending line is number 245:

```c++
#   elif (defined(__clang__) && __has_builtin(__builtin_bswap16))
```

The relevant output from make is

```
Making all in IIDC                                                                      [6/1937]
make[2]: Entering directory '/home/douglass/src/micro-manager/micro-manager/DeviceAdapters/IIDC'
/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_
TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" -DPACKAGE_STRING=\"Micro-Manager\ 1.4\" -DPA
CKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERS
ION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_S
TRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNIS
TD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_BOOST=/\*\*/ -DHAVE_BOOST_THREAD=/\*\*/ -D
HAVE_BOOST_ASIO=/\*\*/ -DHAVE_BOOST_SYSTEM=/\*\*/ -DHAVE_BOOST_CHRONO=/\*\*/ -DHAVE_BOOST_DATE_T
IME=/\*\*/ -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I.  -pthread -I/u
sr/include -DBOOST_THREAD_VERSION=2   -I/home/douglass/src/micro-manager/micro-manager/DeviceAda
pters/../MMDevice -pthread -I/usr/include  -g -O2 -MT libmmgr_dal_IIDC_la-MMIIDCCamera.lo -MD -M
P -MF .deps/libmmgr_dal_IIDC_la-MMIIDCCamera.Tpo -c -o libmmgr_dal_IIDC_la-MMIIDCCamera.lo `test
 -f 'MMIIDCCamera.cpp' || echo './'`MMIIDCCamera.cpp
libtool: compile:  g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPA
CKAGE_VERSION=\"1.4\" "-DPACKAGE_STRING=\"Micro-Manager 1.4\"" -DPACKAGE_BUGREPORT=\"info@micro-
manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 
-DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -
DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_
OBJDIR=\".libs/\" "-DHAVE_BOOST=/**/" "-DHAVE_BOOST_THREAD=/**/" "-DHAVE_BOOST_ASIO=/**/" "-DHAV
E_BOOST_SYSTEM=/**/" "-DHAVE_BOOST_CHRONO=/**/" "-DHAVE_BOOST_DATE_TIME=/**/" -DHAVE__BOOL=1 -DH
AVE_STDBOOL_H=1 -DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I. -pthread -I/usr/include -DBOOST_THREAD_VERS
ION=2 -I/home/douglass/src/micro-manager/micro-manager/DeviceAdapters/../MMDevice -pthread -I/us
r/include -g -O2 -MT libmmgr_dal_IIDC_la-MMIIDCCamera.lo -MD -MP -MF .deps/libmmgr_dal_IIDC_la-M
MIIDCCamera.Tpo -c MMIIDCCamera.cpp  -fPIC -DPIC -o .libs/libmmgr_dal_IIDC_la-MMIIDCCamera.o
MMIIDCCamera.cpp:245:46: error: missing binary operator before token "("
 #   elif (defined(__clang__) && __has_builtin(__builtin_bswap16))
```

# Current solution

I've had success with building mm2 by ignoring errors during `make`, but this is admittedly dangerous since it will ignore all errors.

```sh
make -i
```

## Other attempts at a solution

Based on comments in **MMIIDCCamera.cpp**, I tried using an older version of gcc without success:

```sh
CC="/usr/bin/gcc-4.6" make
```

Interestingly, I tried building on a different Linux x64 system at home and had no problems. The cause of this problem may be related to a specific version of the Linux build tools, but I will have to wait until I go home tonight to check version numbers.

# System info

## Build fails

```sh
uname --all
Linux lebpc18 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux

gcc -dumpversion
4.9.2

make --version
GNU Make 4.0

ldd --version
ldd (Debian GLIBC 2.19-18+deb8u7) 2.19
```

kmdouglass, 2017-01-26:
Well, I guess the problem is not coming from differing glibc versions, because the build that succeeds on my home machine has the same versions as the software that leads to the failed build on my work machine:

```
uname --all
Linux kmd-laptop 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux

gcc -dumpversion
4.9.2

make --version
GNU Make 4.0

ldd --version
ldd (Debian GLIBC 2.19-18+deb8u7) 2.19

```

Is there any way to skip a build for a specific device adapter? I tried modifying some of the Makefiles, but this is almost certainly not the right way to do it...

marktsuchida, 2017-01-27:
The IIDC adapter is only added to the build if libdc1394 is detected, so one question is whether the system on which the build succeeded had libdc1394. You should see a pretty-printed summary of (some) detected libraries when you run `./configure`, which should tell you.

If the answer is no, it may yet be a GCC compatibility issue. I know I didn't test the build on Linux after adding that line (it's intended for OS X where the compiler is clang), so my bad. GCC does not have `__has_builtin` (https://gcc.gnu.org/bugzilla/show_bug.cgi?id=66970). (Note that clang defines `__GNUC__`.)

For the goal of just getting mm2 to build for the time being, you can change the offending line(s) to
```
#   elif 0
```

Line 210 doesn't seem to be giving you an error, probably because `defined(_XOPEN_SOURCE)` is evaluating to true.

There is no way to easily disable a particular device adapter, although I've been thinking about it. You could delete the lines from the end of `DeviceAdapters/configure.ac` and `DeviceAdapters/Makefile.am`, where all the adapters are listed (definitions of `device_adapter_dirs` and `SUBDIRS`, respectively), and rerun `autogen.sh`.

I'm leaving this issue open until we fix the build of IIDC with GCC.

kmdouglass, 2017-01-27:
Hi @marktsuchida ,
Checking for libdc1394 was a good suggestion.

I can confirm that **libdc1394 is present on the machine whose mm2 build is failing.** On my home machine, where the build succeeds, libdc1394 is not present.

csachs, 2017-02-10:
This bug is too small to leave it open for long ;)
I was puzzled why it didn't occur on my Linux build (Ubuntu Xenial) … well it seems gcc added the particular macro in the 5.x line (my default gcc being 5.4)
Clang theirselves suggest adding a `#ifndef` check ( http://clang.llvm.org/docs/LanguageExtensions.html#has-builtin ) I'd suggest the same ( PR #529 ).
With this addition, the build works for g++-4.9, g++-5.4 and clang-3.8 (all under Ubuntu Xenial).

commit 5a3d77fb (2017-02-11): Fixes #516 (__has_builtin not present in pre-5.0 g++ versions) (Christian Sachs)

git-svn-id: https://valelab.ucsf.edu/svn/micromanager2/trunk@16308 d0ab736e-dc22-4aeb-8dc9-08def0aa14fd
closed 2017-02-11 by marktsuchida