micro-manager/micro-manager #520: getLastTaggedImage() returns data from CircularBuffer when the CB is empty

kmdouglass, 2017-01-30:
# Bug description

`mmc.getLastTaggedImage()` returns image data from the circular buffer, even when the buffer is "empty." There are two notable behaviors:

1. `getLastTaggedImage()` is called before starting any acquisitions, including Live feeds. In this case it returns a blank image (all zeros).
2. `getLastTaggedImage()` is called after any acquisition during a MM session. In this case, images from the most recent acquisition are returned, even when `mmc.clearCircularBuffer()` has been explicitly called before `getLastTaggedImage()`.

I believe this is a bug in the sense that the CircularBuffer is not implemented in the way it was intended to be used. In particular, I would expect that it should raise a *Circular buffer is empty* exception in these cases based on its documentation. However, the condition under which it should  raise this exception seems to never be satisfied (see below).

## How to reproduce the bug

The MM2 Beanshell script below will return image data from the CircularBuffer using MMCore's getLastTaggedImage():

```
store = mm.data().createRAMDatastore();
// Create a display to show images as they are acquired.
mm.displays().createDisplay(store);

builder = mm.data().getCoordsBuilder().z(0).channel(0).stagePosition(0);

mmc.clearCircularBuffer();
tagged = mmc.getLastTaggedImage();

// Convert to an Image at the desired timepoint.
image = mm.data().convertTaggedImage(tagged, builder.time(1).build(), null);
store.putImage(image);

mm.displays().manage(store);
```

If this script is run after any acquisition, the Datastore will contain an image from that acqusition, despite the call to `mmc.clearCircularBuffer()`. If it is run before any acquisition has been started, including the Live feed, it will contain a blank image.

## Expected behavior

The expected behavior is that a *Circular buffer is empty* exception will be raised when calling `mmc.getLastTaggedImage()`, similar to calling `mmc.getPopNextTaggedImage()`.

# Possible cause of the problem

I'm assuming that `getLastTaggedImage()` in the MMCoreJ wrapper ultimately results in a call to `CircularBuffer::GetTopImage()`. The code for this method is:

```c++
const unsigned char* CircularBuffer::GetTopImage() const
{
   MMThreadGuard guard(g_bufferLock);

   if (frameArray_.size() == 0)
      return 0;

   if (insertIndex_ == 0)
      return frameArray_[0].GetPixels(0, 0);
   else
      return frameArray_[(insertIndex_-1) % frameArray_.size()].GetPixels(0, 0);
}
```

`frameArray_` is a FrameBuffer instance whose behavior is defined in *MMDevice/ImgBuffer.cpp*. `frameArray_.size()` should return 0 if there has been no data inserted into `frameArray_`. (See, for example, http://www.cplusplus.com/reference/vector/vector/size/) However, there is a call to `frameArray_.Preallocate()` inside `CircularBuffer::Initialize(...)`, which might be filling `frameArray_` with zeros. If the Initialize method is being called anywhere before the above example script is run, then `frameArray_.size() == 0` will never evaluate True.

When `getLastTaggedImage()` returns an image after explicitly calling `mmc.clearCircularBuffer()`, I think it's simply returning the image in `frameArray_[0]`. In particular, the definition of `mmc.clearCircularBuffer()` is

```c++
void CMMCore::clearCircularBuffer() throw (CMMError)
{
   cbuf_->Clear();
}
```

and `CircularBuffer.Clear()`:

```c++
void Clear() {MMThreadGuard guard(g_bufferLock); insertIndex_=0; saveIndex_=0; overflow_ = false;}
```

So `insertIndex_`--which is used by `CircularBuffer::GetTopImage()`--is reset to zero, which results in a call to `frameArray_[0].GetPixels(0, 0)`.

marktsuchida, 2017-01-31:
@kmdouglass Thanks for the detailed report! I believe the commits I made yesterday (actually, without being aware of your report -- a strange coincidence) addresses this issue. I now get an exception when I run
```
mmc.clearCircularBuffer();
mmc.getLastTaggedImage();
```

closed 2017-01-31 by marktsuchida

kmdouglass, 2017-01-31:
Thanks Mark. I'm seeing the exception now when I run the above script.