micro-manager/micro-manager #521: mm2 build for Linux x64 now failing due to recent changes to FrameBuffer.cpp

kmdouglass, 2017-02-02:
Sorry to keep raising Linux build issues, but it looks like GCC doesn't like the new changes to FrameBuffer.cpp:

```
make[2]: Entering directory '/home/douglass/src/micro-manager/micro-manager/MMCore'
depbase=`echo FrameBuffer.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" -DPACKAGE_STRING=\"Micro-Manager\ 1.4\" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_BOOST=/\*\*/ -DHAVE_BOOST_DATE_TIME=/\*\*/ -DHAVE_BOOST_SYSTEM=/\*\*/ -DHAVE_BOOST_THREAD=/\*\*/ -DSTDC_HEADERS=1 -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DHAVE_MEMSET=1 -DHAVE_LIBDL=1 -I.  -pthread -I/usr/include -DBOOST_THREAD_VERSION=2 -DBOOST_THREAD_DONT_PROVIDE_CONDITION   -g -O2 -MT FrameBuffer.lo -MD -MP -MF $depbase.Tpo -c -o FrameBuffer.lo FrameBuffer.cpp &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" "-DPACKAGE_STRING=\"Micro-Manager 1.4\"" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" "-DHAVE_BOOST=/**/" "-DHAVE_BOOST_DATE_TIME=/**/" "-DHAVE_BOOST_SYSTEM=/**/" "-DHAVE_BOOST_THREAD=/**/" -DSTDC_HEADERS=1 -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DHAVE_MEMSET=1 -DHAVE_LIBDL=1 -I. -pthread -I/usr/include -DBOOST_THREAD_VERSION=2 -DBOOST_THREAD_DONT_PROVIDE_CONDITION -g -O2 -MT FrameBuffer.lo -MD -MP -MF .deps/FrameBuffer.Tpo -c FrameBuffer.cpp  -fPIC -DPIC -o .libs/FrameBuffer.o
FrameBuffer.cpp: In constructor 'mm::ImgBuffer::ImgBuffer(unsigned int, unsigned int, unsigned int)':
FrameBuffer.cpp:29:47: error: 'memset' was not declared in this scope
    memset(pixels_, 0, xSize * ySize * pixDepth);
                                               ^
FrameBuffer.cpp: In member function 'void mm::ImgBuffer::SetPixels(const void*)':
FrameBuffer.cpp:44:60: error: 'memcpy' was not declared in this scope
    memcpy((void*)pixels_, pix, width_ * height_ * pixDepth_);
                                                            ^
FrameBuffer.cpp: In member function 'void mm::ImgBuffer::Resize(unsigned int, unsigned int)':
FrameBuffer.cpp:73:51: error: 'memset' was not declared in this scope
    memset(pixels_, 0, width_ * height_ * pixDepth_);
                                                   ^
Makefile:677: recipe for target 'FrameBuffer.lo' failed
make[2]: *** [FrameBuffer.lo] Error 1
make[2]: Leaving directory '/home/douglass/src/micro-manager/micro-manager/MMCore'
```

```
uname -a
Linux xxxx 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux

gcc -dumpversion
4.9.2

make -v
GNU Make 4.0

ldd --version
ldd (Debian GLIBC 2.19-18+deb8u7) 2.19
```

This should probably go along with #516 since they're both GCC related and result in failed builds.

csachs, 2017-02-03:
Having the same problem on Ubuntu with `gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4)` 
 can be fixed via #522.
Probably Travis could be set up to perform CI on the github version?
(And it would be really great if the whole workflow would move to github, instead of merely mirroring a SVN repository ;))

hadim, 2017-02-04:
If it can help I use this script to build mm on my system (Linux Mint) : 

```bash
#!/usr/bin/env sh
set -e

# Script used to build MM from source

# Dependencies (ubuntu / debian)
# sudo apt-get install subversion build-essential autoconf automake libtool pkg-config libboost1.54-all-dev zlib1g-dev swig openjdk-7-jdk ant python-dev python-numpy-dev

CURR_DIR=$(pwd)

if [ ! -f "ij149.zip" ]; then
    wget http://rsb.info.nih.gov/ij/download/zips/ij149.zip
fi

rm -fr ImageJ/
if [ ! -d "ImageJ" ]; then
  unzip ij*.zip
fi

if [ ! -f "ImageJ/MMConfig_demo.cfg" ]; then
  wget https://raw.githubusercontent.com/micro-manager/micro-manager/mm2/bindist/any-platform/MMConfig_demo.cfg -O ImageJ/MMConfig_demo.cfg
fi

if [ ! -d "micro-manager" ]; then
  git clone https://github.com/micro-manager/micro-manager.git
fi

cd micro-manager/
GIT_HASH=$(git rev-parse --short HEAD)

# Build MM
./autogen.sh
CC=gcc CXX=g++ ./configure --enable-imagej-plugin=$CURR_DIR/ImageJ JAVA_HOME=/usr/lib/jvm/default-java/
make fetchdeps
make

make install

cd ../

# Generate zip bundle
mkdir -p bundles/
BUNDLE_NAME="$(date +"%Y.%m.%d.%H.%M").MicroManager-$GIT_HASH.zip"
zip -r bundles/$BUNDLE_NAME ImageJ/
```

It should be easy to adapt it to Travis.