micro-manager/micro-manager #534: MM2: creating a datastore in a for loop fails at start

julou, 2017-03-06:
Hello,

writing an MM2 script (in MM2beta-20170220), I stumbled upon this weird behaviour: 

`if (true) {
store = mm.data().createRAMDatastore(); 
}` returns an empty object when no datastore has been created since MM2 was started (no problem after snapping one image, or running `mm.data().createRAMDatastore()`)...

So far, my workaround is to create a ram datastore and close it at the beginning of the script.
If this is indeed a big, it would probably be worth fixing!

Best,
Thomas

nicost, 2017-03-17:
I can not reproduce that problem.  When I run the script:

`if (true) {
	store = mm.data().createRAMDatastore();
	if (store == null) {
		mm.scripter().message("Store is null");
	} else {
		mm.scripter().message("Found a store");
	}
}`

the output is:
`Found a store`

even right after starting MM and not doing anything else before.

julou, 2017-03-18:
true... it works for me as well!

my case is the following: this doesn't run at startup before a store has been created outside the for loop
```
if (true) {
   store = mm.data().createRAMDatastore();
 }
store.close();
```

the error is 
 ```
Line 4: evaluation error : Sourced file: inline evaluation of: ``if (true) {    store = mm.data().createRAMDatastore();  }  store.close();'' : Attempt to resolve method: close() on undefined variable or class name: store
```

To the contrary, this runs fine:
```
store = mm.data().createRAMDatastore();
store.close();
```

My understanding of datastores being superficial, maybe this is not a bug but an expected behaviour....

nicost, 2017-03-18:
I am not completely sure about the scoping rules in Beanshell, but most likely, ```store``` is out of scope in your first example, hence it is undefined (as the error message states).  Put it before the closing bracket and it will be fine. Most serious programming languages behave this way.  

This syntax is completely independent of Micro-Manager.  You will get the same error when using the Java String object:
```
if (true) {
	test = "Test";
	//test.isEmpty();
}
test.isEmpty();
```

julou, 2017-03-18:
makes full sense… thx a lot!

closed 2017-03-18 by julou