micro-manager/micro-manager #545: Allow users more flexibility in controlling filenames acquired via the MDA

kmdouglass, 2017-03-20:
Currently in MM2, there is no way to control whether an integer is appended to the filenames containing data acquired via a multi-dimensional acquisition (MDA) and that are subsequently saved to disk; a unique integer identifier is always included in the filenames. The problem is that this behavior limits the flexibility of users to control the names of their data ([See this post on the mailing list, for example](https://sourceforge.net/p/micro-manager/mailman/message/35734746/).)

If the developers and community agree that this behavior is not desirable, then there are two possibilities that I can see to address this:

## Allow users to control whether the integer is added

The identifier is currently appended in the method `createAcqDirectory()`, which is always called in the constructor of `org.micromanager.acquisition.internal.MMAcquisition` [(link)](https://github.com/micro-manager/micro-manager/blob/mm2/mmstudio/src/main/java/org/micromanager/acquisition/internal/MMAcquisition.java). One could allow for a boolean flag passed to the constructor that toggles whether this method is called.

If this approach is taken, we would need to decide what to do in the case that the folder already exists and the user has chosen not to create a unique directory. This approach would also probably require updating the MDA GUI to include a checkbox (or similar widget) to set this flag.

## Modify `createAcqDirectory()` to use `mm.data().getUniqueSaveDirectory()` instead

This approach would at least make `createAcqDirectory()` more consistent with user scripts that use `getUniqueSaveDirectory()`. The behavior of `getUniqueSaveDirectory()` only appends an integer to the directory name **if a folder of the same name already exists**.

The downside to this approach is that users have lived with the current behavior for a long time and may be surprised when the MDA stops automatically adding an integer starting at 1 to their data. The other downside is that it doesn't fully give users flexibility in naming their datasets; an integer will still be appended to the folder and filenames if the directory already exists.

-----

What are other people's thoughts on this matter? I personally prefer option 2 above because it's easy to implement and improves consistency throughout the MM code base.

kmdouglass, 2017-03-25:
#552 should be reviewed and merged before implementing the second option discussed above if it's decided to make the MDA use `getUniqueSaveDirectory()`.

marktsuchida, 2020-08-31:
This still seems relevant. Script writers should have a way to have total control over file naming. Implementing #266 should provide that capability, so closing this for now.

closed 2020-08-31 by marktsuchida