micro-manager/micro-manager #548: Coords.CHANNEL not equivalent to Coords.CHANNEL_SHORT

nicost, 2017-03-22:
For a datastore with 2 channels:
datastore.getAxisLength(Coords.CHANNEL) != datastore.getAxisLength(Coords.CHANNEL_SHORT).

datastore.getAxisLength(Coords.CHANNEL) returns 2 (correct)
datastore.getAxisLength(Coords.CHANNEL_SHORT) returns 0 (incorrect).

Is this indeed a bug, or is there some intention that I do not understand?

marktsuchida, 2017-03-22:
Not a bug, I don't think, but definitely a source of potential confusion. The `_SHORT` versions are for use with the `DefaultCoords.fromNormalizedString` method, but then I'm not sure why the `_SHORT` final fields need to be exposed in the API. We should probably remove it if we can confirm that that is the only way it is used.

nicost, 2017-03-23:
Looks to me that those _SHORT versions should indeed not be exposed if they are not intended to be used by outside code.

nicost, 2020-01-31:
Looks like those are no longer present

closed 2020-01-31 by nicost