micro-manager/micro-manager #554: Chain icon in inspector window next color button does not honor color changes

nicost, 2017-03-29:
Linking viewer windows using this "chain" function causes changes in brightness and contrast to be propagated to all viewers, but not changes in the color that is selected through the icon right next to it.  I am not sure what the most useful behavior is, but the current behavior is certainly unexpected since the placement of this icon suggests that it links the colors (and possibly things that are further away as well).  If the current behavior is maintained, it may be better to place the icon under the histogram, to the right of the min/max/mean section.  It could also make sense to replace the individual chain icons for all channels with a single one that links the brightness/color and display mode of all channels (and would be placed next to the "Color mode" pull down on top of the histograms.

marktsuchida, 2017-03-29:
I kind of like your latter suggestion for its simplicity, but I can think of cases where somebody might want to link one channel but not another (maybe the fluorescent channels are quantitative but the bright field should be adjusted per dataset).
So I'll plan on moving it to the right of the Min/Max/Mean/Std (with some minimal gap).
Hopefully this isn't urgent since the popup menu does explain what it does.

nicost, 2020-04-03:
The chain icon is now under the histogram, to the right.

closed 2020-04-03 by nicost