micro-manager/micro-manager #556: Wrong stage position name in ome xml

guiwitz, 2017-03-31:
The stage position name (typically Pos0, Pos1 etc.) saved in the ome-xml of MM2 datasets is wrong. The method getStageLabelName reports the name of the first position (typically Pos0) for all positions. This isn't the case for MM1.4 datasets where the correct position names are reported. The following Fiji bsh code which prints out all stage position names for a dataset exposes the problem:

//Getting all stage position names for MM dataset
import loci.formats.*;
import loci.formats.ome.OMEXMLMetadata;
import loci.formats.in.OMETiffReader;

ImageReader reader = new ImageReader();
ClassList enabledClasses = new ClassList(IFormatReader.class);
enabledClasses.addClass(loci.formats.in.OMETiffReader.class);
ImageReader reader = new ImageReader(enabledClasses);
OMEXMLMetadata omeMeta = MetadataTools.createOMEXMLMetadata();
reader.setMetadataStore(omeMeta);
reader.setId("/whatever/the/path/to/your/file/is/myfile_MMStack.ome.tif");
seriesCount = reader.getSeriesCount();
reader.close();
for (int i=0; i<seriesCount; i++) {
	cur_pos = omeMeta.getStageLabelName(i);
	print(cur_pos);
}
//

nicost, 2017-03-31:
I tried to reproduce this by generating a PositionList with 3 positions (labeled Pos0, Pos1, Pos2) using the demo setup and saving it.  Looking at the generated xml (copied below) all looks fine.  Can you generate an example that is easy to reproduce and that does not rely on non-Micro-Manager code?

```
{
   "VERSION": 3,
   "ID": "Micro-Manager XY-position list",
   "POSITIONS": [
      {
         "GRID_COL": 0,
         "DEVICES": [
            {
               "DEVICE": "Z",
               "AXES": 1,
               "Y": 0,
               "X": 0,
               "Z": 0
            },
            {
               "DEVICE": "XY",
               "AXES": 2,
               "Y": -0,
               "X": -0,
               "Z": 0
            }
         ],
         "PROPERTIES": {},
         "DEFAULT_Z_STAGE": "Z",
         "LABEL": "Pos0",
         "GRID_ROW": 0,
         "DEFAULT_XY_STAGE": "XY"
      },
      {
         "GRID_COL": 0,
         "DEVICES": [
            {
               "DEVICE": "Z",
               "AXES": 1,
               "Y": 0,
               "X": 0,
               "Z": 0
            },
            {
               "DEVICE": "XY",
               "AXES": 2,
               "Y": -0,
               "X": -0,
               "Z": 0
            }
         ],
         "PROPERTIES": {},
         "DEFAULT_Z_STAGE": "Z",
         "LABEL": "Pos1",
         "GRID_ROW": 0,
         "DEFAULT_XY_STAGE": "XY"
      }, {
         "GRID_COL": 0,
         "DEVICES": [
            {
               "DEVICE": "Z",
               "AXES": 1,
               "Y": 0,
               "X": 0,
               "Z": 0
            },
            {
               "DEVICE": "XY",
               "AXES": 2,
               "Y": -0,
               "X": -0,
               "Z": 0
            }
         ],
         "PROPERTIES": {},
         "DEFAULT_Z_STAGE": "Z",
         "LABEL": "Pos2",
         "GRID_ROW": 0,
         "DEFAULT_XY_STAGE": "XY"
      }
   ]
}

```

nicost, 2017-03-31:
Wait, you are not talking about the PositionList but the ome metadata saved with images. OK, forget what I wrote above...

nicost, 2017-03-31:
Fixed in PR #557

closed 2017-03-31 by nicost