micro-manager/micro-manager #561: Groups and presents unavailable - error when trying to add them

npyoung, 2017-04-13:
Adding a group with the + button results in the follow error dialog:

```Error: org.micromanager.internals.utils.PropertyTableData.update(PropertyTableData.java:235)```

The Group Editor window opens, but no device properties are listed. However, when opening the Device Property Browser, all configured properties show up as expected, and the microscope can be controlled. It's just groups and presets that don't work.

This occurs when making a new configuration from scratch, or when loading up a config made in MM 1.4. In the later case, the existing groups and presets do not appear under "Configuration settings" on the main window.

OS: Windows 7
MM: v2.0.0-beta2

nicost, 2017-04-17:
Can you try the same in the latest nightly build?  Also good to test with the demo configuration (if the problem persists in the nightly build).

npyoung, 2017-04-29:
This was the latest nightly build as of the date of posting - 4acd302 was the only change since then.

Problem persists using the demo config too.

nicost, 2017-04-29:
I can not reproduce this problem.  Generate a Problem Report and look for Java exceptions.  Copy the whole Exception chain, hopefully it provides a clue.

npyoung, 2017-05-26:
Since trying to reproduce this with the latest nightlies (5/25), the bug seems to have disappeared.

closed 2017-05-26 by npyoung