micro-manager/micro-manager #570: HCS site generator calibration does not get applied

npyoung, 2017-05-26:
After carrying out the steps in the HCS Site Generator's calibration GUI, calibration does not seem to take place. If, after calibrating, I go into "Move Stage" mode and select a location, the stage doesn't move to that location - it retains any offset it had before calibration. This is true if single point (i.e. move to the center of the given well) calibration is used, or if 4 point (top/right/bottom/left of well) calibration is used.

I noticed that after centering over a well and calibrating, the red square indicating where the software thinks the stage is positioned is not centered in the well, indicating to me immediately that the calibration didn't work. It seems like the calibration offset is not being saved or used.

Hardware: Leica DMI 6000
OS: Windows 7 64bit
uM: 25 May 2017 nightly
HCS Site Generator: v1.5.0

We also had this issue in uM 1.4 on the same rig.

nicost, 2017-05-30:
Can you replicate this with the demo configuration?  
I can not and therefore expect that you ran into an issue with your hardware.  Create a problem report and see if there are errors.  If you send it, I'll have a quick look at it.

npyoung, 2017-06-07:
HCS calibration appears to work in demo mode (although it's hard to tell without a physical stage). If I move to a position offset from the center of a well and then calibrate (single point) as if it were in the center, the red square location marker jumps to the center of the well after calibration.

I'm convinced this is a software issue. The stage shouldn't have to move at all when the calibrate button is pressed. All that should happen is the stage position gets read and the offset applied in software. I get a "XY Stage set at position: ..." popup after calibrating, which tells me the stage position is being read, but not being used properly in software.

I just sent in a problem report. Hopefully you can see something helpful.

nicost, 2017-06-08:
Regretfully, nothing pops out in the problem report.  This absolutely is a software issue, but needs your hardware to be reproduced....  I do not see this problem with ASI stages, and do not have access to Leica DMI.  It would be great if you can do some further debugging on your setup.

nicost, 2020-02-02:
Close for lack of updates

closed 2020-02-02 by nicost