micro-manager/micro-manager #573: [mm2] MMStudio fails to build on Linux

Armavica, 2017-07-06:
I am trying to compile mm2 on Linux, but MMStudio fails to build, the error seems to be that the file `org.scijava.plugin.Plugin` cannot be found. Do you have an idea why this happens? I attach what seems relevant, please let me know if you need more.

- [`./configure`](https://paste.fulltxt.net/-1VfePSMBJgLl)
- `make fetchdeps` reports success
- [`make`](https://paste.fulltxt.net/JO@_QnAJP)

kmdouglass, 2017-07-07:
Did you download the 3rd party libraries from the `3rdpartypublic` SVN repository and place the folder in the same directory as the one containing the mm2 source code folder?

See [this section](https://github.com/micro-manager/micro-manager/blob/mm2/doc/how-to-build.md#obtaining-the-source-code) of the build directions.

Armavica, 2017-07-07:
Yes, I did.

cErikson, 2019-02-20:
I am also having this problem. I am building with the arch with the commit 88ab1ff051b548aafced024c46f902720eb9d553.

fetchdeps reports:
`[ivy:resolve] 	found org.scijava#scijava-common;2.46.0 in central`

cErikson, 2019-03-09:
I managed to resolve this problem by changing the default compiler from OpenJDK10 to OpenJDK8

dedeibel, 2019-04-10:
I can recommend this build script for linux: https://github.com/hadim/mm_scripts/blob/master/build-mm.sh

closed 2020-04-05 by nicost