micro-manager/micro-manager #599: MMCorePy wrapping of SLM related methods (unsigned char * SWIG input typemap).

zindy, 2017-12-12:
Hi everyone, thought I would share what I've been up to.

Two things I'm trying to wrap (robustly):

* `std::vector<unsigned char *>` as in input type
* `unsigned char *` as an input type

The first one seems OK with the following code:
```cpp
/* tell SWIG to treat char ** as a list of strings */
/* From https://stackoverflow.com/questions/3494598/passing-a-list-of-strings-to-from-python-ctypes-to-c-function-expecting-char */
/* XXX No need to freearg the vector, right? */
%typemap(in) std::vector<unsigned char *> {
    // check if is a list
    if(PyList_Check($input))
    {
        long expectedLength = (arg1)->getSLMWidth(arg2) * (arg1)->getSLMHeight(arg2);

        Py_ssize_t size = PyList_Size($input);
        std::vector<unsigned char*> inputVector;

        for(Py_ssize_t i = 0; i < size; i++)
        {
            printf("Pushing %d\n",  i);
            PyObject * o = PyList_GetItem($input, i);
            if(PyString_Check(o))
            {
                if (PyString_Size(o) != expectedLength)
                {
                    PyErr_SetString(PyExc_TypeError, "One of the Image strings is the wrong length for this SLM.");
                    return NULL;
                }
   
                inputVector.push_back((unsigned char *)PyString_AsString(o));
            }
            else
            {
                PyErr_SetString(PyExc_TypeError, "list must contain strings");
                return NULL;
            }
        }
        $1 = inputVector;
    }
    else
    {
        PyErr_SetString(PyExc_TypeError, "not a list");
        return NULL;
    }
}
```

However, I haven't had much success with `unsigned char *`. Sure, you can use the nuclear approach for now:
```cpp
// input arguments
%apply char* {unsigned char *};
```

But I'd like something a bit more sophisticated (robust), which at least checks the string length. I thought the following code would work:

```cpp
%typemap(in) unsigned char *
{
   // Assume that we are sending an image to an SLM device, one byte per pixel (monochrome grayscale).
   if(!PyString_Check($input))
   {
      PyErr_SetString(PyExc_TypeError, "not a string");
      return NULL;
   }

   long expectedLength = (arg1)->getSLMWidth(arg2) * (arg1)->getSLMHeight(arg2);
   long receivedLength = PyString_Size($input);
   printf("expected: %d -- received: %d\n",expectedLength,receivedLength);

   if (receivedLength != expectedLength)
   {
      PyErr_SetString(PyExc_TypeError, "Image dimensions are wrong for this SLM.");
      return NULL;
   }
   
   unsigned char *uchar_ptr = (unsigned char *)PyString_AsString($input);
   $1 = uchar_ptr;
}
```
But.... no, this doesn't help. Passing two strings (devname and val) to ```mmc.setSLMImage(devname,val)``` leads to:

"""
Traceback (most recent call last):
  File "test_slm.py", line 44, in <module>    mmc.setSLMImage(devname,val)
  File "C:\Program Files\Micro-Manager-1.4\MMCorePy.py", line 6537, in setSLMImage
    return _MMCorePy.CMMCore_setSLMImage(self, *args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'CMMCore_setSLMImage'.
  Possible C/C++ prototypes are:
    CMMCore::setSLMImage(char const *,unsigned char *)
    CMMCore::setSLMImage(char const *,imgRGB32)
"""

And in the SWIG generated cxx, I have a failure at:
"""
        int res = SWIG_ConvertPtr(argv[2], &vptr, SWIGTYPE_p_unsigned_char, 0);
        _v = SWIG_CheckState(res);
"""
I traced this back to ```SWIG_Python_ConvertPtrAndOwn(PyObject *obj, void **ptr, swig_type_info *ty, int flags, int *own)```, and *possibly* the pointer conversion is failing for the lack of a freearg typemap. What do I need to free, though? Something along the lines of this?

```
%typemap(freearg) unsigned char * {
   free($1);
}
```

But wouldn't the block of ```unsigned char *``` memory pointed at by ```PyString_AsString()``` be freed when the string is deleted from Python? Or is that wishful thinking?

Although.... the underlying issue is the use unsigned char * in MMCore.cpp, which SWIG can't wrap from strings without meddling. It would make the SWIG wrapping so much easier if these few methods that use unsigned char * would use signed char * instead.

So if the fix to this isn't obvious, would it be a major disruption if I took a stab at changing the argument types in **MMCore.cpp**? As far as I can tell, this would only affect the few SLM related methods, but I would make sure both the Java and Python wrappers work as expected before submitting any changes.

Cheers,
Egor

nicost, 2017-12-12:
I do not particularly like to use char* instead of unsigned char*.  A char* in C implies a null terminated string, whereas that is clearly not what we have here (a pointer to a block of memory).  I agree that the MMCore interface is not very clean, I think that the SLM interface was copied from the image-related code.  There are no checks on size of the memory that is passed, or on its type (which can be 1 or 2 bytes per value).  Those values are established using the separate GetSLMWidth/Height/BytesPerPixel/NumberOfComponents functions, and the hope is that everyone agrees to those numbers (or bad things will happen).  However, that memory is not a null terminated string, it is a block of memory containing unsigned numbers that are either 1 or 2 bytes in length.  I don't know anything about Python and the types it can deal with, but a signed char seems wrong.

zindy, 2017-12-13:
Hi Nico,

I totally agree that in this case, unsigned char * seems more appropriate for an 8 bit greyscale image.

So back to the drawing board, the following code seems to work for now. Here we have a conversion from a Python string to char * + int size, followed by an extension of the CMMCore class (in the .i SWIG interface)

```cpp
%rename(setSLMImage) setSLMImage_pywrap;
%apply (char *STRING, int LENGTH) { (char *pixels, int receivedLength) };
%extend CMMCore {
void setSLMImage_pywrap(const char* slmLabel, char *pixels, int receivedLength)
{
   long expectedLength = self->getSLMWidth(slmLabel) * self->getSLMHeight(slmLabel);
   //printf("expected: %d -- received: %d\n",expectedLength,receivedLength);

   if (receivedLength != expectedLength)
   {
      PyErr_SetString(PyExc_TypeError, "Image dimensions are wrong for this SLM.");
   }
   self->setSLMImage(slmLabel, (unsigned char *)pixels);
}
}
```

The code is not elegant, but will check that the right number of strings is passed to ```mmc.setSLMImage(slmLabel,pixels)``` and that the *pixels* string is the right size.

I'll check later if this works in Windows (for now, I only tested the code in Linux).

And if anyone has a better suggestion, I'm all ears :-)

nicost, 2018-01-26:
I think this is all in and working now?  I'll close this but please re-open if I missed something

closed 2018-01-26 by nicost