micro-manager/micro-manager #602: micro-manager 2.0 build failure in FakeCamera.cpp

ngladitz, 2018-01-16:
My Linux build of the mm2 branch fails with:
```
FakeCamera.cpp:658:63: error: no member named 'imread' in namespace 'cv'
        cv::Mat img = path == lastFailedPath_ ? lastFailedImg_ : cv::imread(path, cv::IMREAD_ANYDEPTH | (color_ ? cv::IMREAD_COLOR : cv::IMREAD_GRAYSCALE));
```

This fixes it for me:
```patch
--- a/DeviceAdapters/FakeCamera/FakeCamera.h
+++ b/DeviceAdapters/FakeCamera/FakeCamera.h
@@ -27,7 +27,7 @@
 
 #include "DeviceBase.h"
 
-#include "opencv/highgui.h"
+#include <opencv/cv.hpp>
 
 #define ERR_INVALID_DEVICE_NAME 10000
 #define OUT_OF_RANGE 10001
```

nicost, 2018-01-26:
Builds that way on my Mac as well.  Not sure about Windows, so put ifdefs around it so that our Windows build will not break.  Committed in the 1.4 branch (commit 16604), will make its way to 2.0 eventually.

closed 2018-01-26 by nicost

ngladitz, 2018-01-26:
Alright thank for trying. Might help to know what the failure was though.
For future reference I was building on Ubuntu 17.10 with a libopencv-dev (3.1.0+dfsg1-1~exp1ubuntu3) package installation.

nicost, 2018-01-26:
Failure was:
In file included from FakeCamera.cpp:24:
In file included from ./FakeCamera.h:33:
<http://mmmacbuild.local:8080/job/Micro-Manager%20OS%20X%20SVN%20build/ws/../MMDependencies/dependencies/include/opencv/cv.hpp>:50:10: error: 'cv.h' file not found with <angled> include; use "quotes" instead
#include <cv.h>
         ^~~~~~
         "cv.h"
In file included from FakeCamera.cpp:24:
In file included from ./FakeCamera.h:33:
In file included from <http://mmmacbuild.local:8080/job/Micro-Manager%20OS%20X%20SVN%20build/ws/../MMDependencies/dependencies/include/opencv/cv.hpp>:50:
<http://mmmacbuild.local:8080/job/Micro-Manager%20OS%20X%20SVN%20build/ws/../MMDependencies/dependencies/include/opencv/cv.h>:67:10: fatal error: 'opencv2/video/tracking.hpp' file not found
#include "opencv2/video/tracking.hpp"