micro-manager/micro-manager #616: `error: use of undeclared identifier 'htonl'` in WineckeSinske.cpp` when trying to build locally

beuerle, 2018-03-08:
I get the following error when trying to build `micro-manager` on linux:

```bash
$ make
[..]
make[2]: Entering directory '/home/bb/tmp/mm/micro-manager/DeviceAdapters/WieneckeSinske'
/bin/sh ../libtool  --tag=CXX   --mode=compile clang++ -DPACKAGE_NAME=\"Micro-Manager\"
-DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" -DPACKAGE_STRING=\"Micro-Manager\
1.4\" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\"
-DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1
-DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1
-DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_BOOST=/\*\*/
-DHAVE_BOOST_THREAD=/\*\*/ -DHAVE_BOOST_ASIO=/\*\*/ -DHAVE_BOOST_SYSTEM=/\*\*/
-DHAVE_BOOST_CHRONO=/\*\*/ -DHAVE_BOOST_DATE_TIME=/\*\*/ -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1
-DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I.    -I/home/bb/tmp/mm/micro-manager/DeviceAdapters/../MMDevice
-pthread -I/usr/include -g -O2 -MT WieneckeSinske.lo -MD -MP -MF .deps/WieneckeSinske.Tpo -c -o
WieneckeSinske.lo WieneckeSinske.cpp
libtool: compile:  clang++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\"
-DPACKAGE_VERSION=\"1.4\" "-DPACKAGE_STRING=\"Micro-Manager 1.4\""
-DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\"
-DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1
-DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1
-DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" "-DHAVE_BOOST=/**/"
"-DHAVE_BOOST_THREAD=/**/" "-DHAVE_BOOST_ASIO=/**/" "-DHAVE_BOOST_SYSTEM=/**/"
"-DHAVE_BOOST_CHRONO=/**/" "-DHAVE_BOOST_DATE_TIME=/**/" -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1
-DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I. -I/home/bb/tmp/mm/micro-manager/DeviceAdapters/../MMDevice
-pthread -I/usr/include -g -O2 -MT WieneckeSinske.lo -MD -MP -MF .deps/WieneckeSinske.Tpo -c
WieneckeSinske.cpp  -fPIC -DPIC -o .libs/WieneckeSinske.o
WieneckeSinske.cpp:241:13: error: use of undeclared identifier 'htonl'
        long tmp = htonl(position);
                   ^
WieneckeSinske.cpp:258:13: error: use of undeclared identifier 'htonl'
        long tmp = htonl(position);
                   ^
WieneckeSinske.cpp:401:13: error: use of undeclared identifier 'htonl'
        long tmp = htonl(velocity);
                   ^
WieneckeSinske.cpp:415:13: error: use of undeclared identifier 'htonl'
        long tmp = htonl(acceleration);
                   ^
4 errors generated.
make[2]: *** [Makefile:526: WieneckeSinske.lo] Error 1
make[2]: Leaving directory '/home/bb/tmp/mm/micro-manager/DeviceAdapters/WieneckeSinske'
make[1]: *** [Makefile:621: all-recursive] Error 1
make[1]: Leaving directory '/home/bb/tmp/mm/micro-manager/DeviceAdapters'
make: *** [Makefile:518: all-recursive] Error 1
```

That's the output of `./configure`:

```bash
$ ./configure
[..]

## ---------------------------- ##
## DeviceAdapters configuration ##
## ---------------------------- ##

Note: This is an incomplete list currently under development.

    Build with FreeImagePlus:               no
    Build with HIDAPI:                      no
    Build with Modbus:                      no
    Build with libdc1394:                   yes
    Build with libgphoto2:                  yes
    Build with libusb 0.1 (libusb-compat):  yes
    Build with OpenCV:                      no

## --------------------------- ##
## Micro-Manager configuration ##
## --------------------------- ##

   PYTHON          = /bin/python
   PYTHON_CPPFLAGS =  -I/usr/include/python3.6m
   NUMPY_CPPFLAGS  =  -I/usr/lib/python3.6/site-packages/numpy/core/include

   Build MMCoreJ (MMCore Java wrapper):    no (Java not available)
   Build MMCorePy (MMCore Python wrapper): yes 
   Build Java application (MMStudio):      no (cannot build MMCoreJ)
   Build device adapters:                  yes
   Build closed-source device adapters:    no
```
`clang++` version:

```bash
$  clang++ --version
clang version 5.0.1 (tags/RELEASE_501/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /bin
```

Any ideas/hints?

S3LSteffen, 2018-03-08:
On Linux htonl seems to be defined only for int32. Can you please try to replace
      long tmp = htonl(xxx);
by 
      int tmp = htonl(xxx);
This could help...

If not:  The same function is used also in DeviceAdapters\ZeissCAN29 and DeviceAdapters\TetheredCam. Do they compile on your system?

beuerle, 2018-03-12:
At the end I commented out the `WieneckeSinske` target:

```diff
diff --git a/DeviceAdapters/Makefile.am b/DeviceAdapters/Makefile.am
index 7c0863570..fe11309d6 100644
--- a/DeviceAdapters/Makefile.am
+++ b/DeviceAdapters/Makefile.am
@@ -166,7 +166,6 @@ SUBDIRS = \
        VariLC \
        Vincent \
        Vortran \
-       WieneckeSinske \
        XCite120PC_Exacte \
        XCiteLed \
        XCiteXT600 \
```

Unfortunately the camera we are using is not supported by micro-manager (we use a ToupTek camer [0]), so I will not use micro-manager (at least for now) and will not investigate this problem further.

Another off-topic comment: `mmimagej` is nearly impossible to use on my system, with the menu raiders more or less inaccessible. This is of course related to my system, but it would be nice overall to have a user interface which is up-to-date.

[0] http://www.touptek.com/product/showproduct.php?lang=en&id=273

closed 2018-03-12 by beuerle

kmdouglass, 2018-03-27:
I believe this is now fixed with a patch in #613