micro-manager/micro-manager #617: MMConfig_demo.cfg file is not installed with Linux builds

kmdouglass, 2018-03-15:
When building Micro-Manager on Linux and running it for the first time, MM expects to find the **MMConfig_demo.cfg** file in the same directory where it was installed. The file however is not present, probably because it is not included in the GitHub repository.

I can personally confirm this bug and it has been reproduced by others: http://micro-manager.3463995.n2.nabble.com/compilation-failures-on-Debian-Testing-tc7589260.html#a7589267

marktsuchida, 2018-03-27:
Classic single-character bug.... Fixed in 1.4, will merge into mm2 shortly.

closed 2018-03-27 by marktsuchida

calocedrus, 2020-08-27:
Has this been merged into mm2? I've just built mm2 (linux ubuntu 18.04) but starting mm shows an error message
`"failed to load hardware configuration ... /MMConfig_demo.cfg": File open failed."`

The file is in `master/Tutorial/Tutorial4`. Are tutorials supposed to be installed alongside the build?

marktsuchida, 2020-08-28:
@calocedrus The demo config file that should be installed is at `bindist/any-platform` in the source tree, not in the (potentially outdated) tutorials. I think line 1 of `bindist/Makefile.am` should cause `MMConfig_demo.cfg` to be installed in the data directory (under `/usr/local/share` if not using `--enable-imagej-plugin`).

Perhaps your settings remember an incorrect path for the config file. You could try clearing `~/.config/Micro-Manager`, or just select the correct path before continuing at the startup dialog.

calocedrus, 2020-09-01:
@marktsuchida Thanks. I had built MM with imageJ (with the configure option `--enable-imagej-plugin`). I've reinstalled/built everything (as I had other issues that I was trying to solve as well) and now the default configuration file loads without complaining. I'm not sure what was wrong.
Note: when MM is built as an ImageJ plugin, `MMConfig_demo.cfg` seems to load from the source tree `micro-manager/bindist/any-platform` (from `git clone https://github.com/micro-manager/micro-manager.git`) .

marktsuchida, 2020-09-01:
@calocedrus Thanks for the update.