micro-manager/micro-manager #619: Confusing behavior when adding new images to Datastores

henrypinkard, 2018-03-17:
When constructing a Datastore that only varies along a single axis (e.g. a single position, single single slice, single channel, time lapse), the Datastore will allow you to add images with Coords that only have one axis specified, but it won't correctly display them. This is because the HashMap that stores images based on their Coords does not consider Z0, P0, C0, T1 equal to Coords with only T1. So the following code won't work:

  Coords.CoordsBuilder builder = studio.data().getCoordsBuilder();
  builder.time(1);
  datastore_.putImage(images.get(0).copyAtCoords(builder.build()));

But this code will:

  Coords.CoordsBuilder builder = studio.data().getCoordsBuilder();
  builder.time(1);
  builder.channel(0);
  builder.stagePosition(0);
  builder.z(0);
  datastore_.putImage(images.get(0).copyAtCoords(builder.build()));

Doesn't seem like this behavior is incorrect, but perhaps Datastore.putImage could more strictly enforce that the Coords it is passed be fully specified?

nicost, 2018-03-18:
Completely agree.  In fact, one can make the argument that the behavior that is enforced by the implementation is incorrect, since one can add an arbitrary number of axes, which should (or should not?) also be set to 0.  The requirement that all axes are set to 0 is scattered throughout the code, so it will be a lot of work to fix (I looked at this a while ago).  CoordsBuilder could simply add the missing axes itself (which is more a work-around than an improved implementation).  

B.t.w., at the moment, development is happening in Mark's mm2-next branch, will will hopefully soon be released as beta4, and will have a lot of the changes I made in 2.0 gamma.

henrypinkard, 2018-03-19:
Good to know. 

What is the best branch/workflow for committing new device adapters?

nicost, 2018-03-19:
DeviceAdapters still first go into the subversion repository (used to build 1.4), and from there propagate to the various 2.0 branches.  Not sure about your write permissions allow it, but if they do, you can put the code in directly.  Alternatively, a PR here is also fine, although it may take a while before those are looked at.

PR #1178 (2021-04-13): Coords cleanup
This PR deals with the long standing issue that Coords with an axis set to zero are not identical to Coords that lack the axis altogether.  I know that Mark though differently about this, but to me, this made no logical sense for a system that is supposed to handle arbitrary axes.  The CoordsBuilder now drops all axes that were set to zero, asking for the value of a missing axis in a Coord will return zero (rather than -1), and the code was updated to work with this new scheme.

In addition, missing images are again filled in.  It is still desirable to add a user interface so th [... cut]
closed 2021-04-13 by nicost