micro-manager/micro-manager #632: controls for multiple Z stages in MM2

jondaniels, 2018-06-08:
I am often trying to control multiple Z stages and wonder if the stage control plugin could expand.  I have working (somewhat hacky) code that works for up to 4 Z stages (the window shows as many as controls as there are Z stages, up to 4).   I'm looking for feedback on whether this is approach is a good one or if there is a better approach.

Here is a screenshot from my current code: 

[image]

It might be convenient to have separate windows for separate stages.  I understand plugins are essentially singletons but maybe a single plugin could spawn other windows.  A separate plugin might be better and isn't limited to an arbitrary number of Z stages.

What do others think?  Especially @marktsuchida and @nicost.

marktsuchida, 2018-06-08:
@jondaniels I certainly like the idea of displaying multiple Z stages side-by-side in the Stage Control window.

The typical maximum number of Z stages is 3 (motorized, piezo, and hardware focus-maintaining ("autofocus") device). I don't think it is a problem if there is an arbitrary limit of 4 stages (if anybody has more, they probably need a fundamentally different interface anyway).

It might be nice if the user can control how many Z stages to show (+ and - buttons?).

jondaniels, 2018-06-08:
@marktsuchida Thanks for the feedback!  I'm looking into incorporating + and - buttons, that's a good idea.

How would you determine how many Z stage controls to show by default? (before clicking +/- buttons)

My current thought is to restore the same number of Z controls the user previously had on a per-configuration basis.  I could store a profile key including the config name with the value (kind of like how the button deltas are stored with a profile key including the stage name).  If the user loads a not-yet-seen configuration name then the value default to number of loaded Z stages.  Does this seem like a good approach or would you suggest a different tack?

marktsuchida, 2018-06-11:
@jondaniels Sounds good to me. I think that's about as good as we can do without major changes to how config files are managed.

jondaniels, 2018-06-14:
@marktsuchida Any possibility of an API call to get the name of the currently-loaded configuration file?  I'm taking a back-door approach.  This is used to remember the number of Z stage controls to show on a per-configuration basis.

I have added the +/- buttons and I feel that this is ready for merging, I have a PR into Nico's branch (as I understand the preferred route at the moment) at https://github.com/nicost/micro-manager/pull/4

jondaniels, 2018-06-20:
Nico merged the PR into his ViewerPlusCV branch so it should make it into 2.0 eventually.

closed 2018-06-20 by jondaniels

hassibrustemi, 2019-10-17:
would you be able to explain how you got the sliders for both z stages? I'm trying to do the same thing and I'm having no luck. Is there a way to do it without writing a custom code? and if not, would you be willing to share your code?

nicost, 2019-10-17:
Code is here: https://github.com/nicost/micro-manager/blob/ViewerPlusCV/mmstudio/src/main/java/org/micromanager/internal/dialogs/StageControlFrame.java