micro-manager/micro-manager #633: Include Python3 wrapper in the MM2 distribution

oeway, 2018-06-09:
Hi, since most python projects have been moved to python3, could you please include a python3 wrapper in the future [distribution of micro-manager](https://micro-manager.org/wiki/Version_2.0)? 

Building MM ourselves can be a pain, especially when we want to target users on different platforms, I think it would be much easier if this can be incorporate in to your building pipeline. 

Thanks a lot.

bryantChhun, 2018-06-19:
I've seen this wrapper here:  https://github.com/zfphil/micro-manager-py36
But it certainly doesn't work on MacOS.  It's implied in the repo that it's for windows, but haven't tested this yet myself.

oeway, 2018-06-20:
@bryantChhun thanks for the pointer, that repo contains a binary file compiled on windows, so it won’t work on Mac.

Also notice that this issue is not about how we can find one wrapper which works for Python3. It’s about a suggestion that MM2 distribution should include Py3 wrapper. 

Without that, everyone wants to use need to compile it or download from elsewhere, thus it’s not easy to distribute a tool depends on that; everyone who provide such wrapper will need to maintain it and recompile it when MM changes, this will not sustain.

aaristov, 2018-06-20:
Great idea, mm2 even in beta is already absolete with python support.

bryantChhun, 2018-06-20:
@jondaniels 
That's my guess but I haven't tried it yet.

@oeway 
I totally agree that all mm2 distributions should include functional Py3 wrappers.  Perhaps this thread will be the beginning of that endeavor :-)

I have, however, finally got the linked wrapper above to work just fine on mm2.  I ran into a problem with "Incompatible device interface version (required = 67; found = 68)", but rolled back to a 2017 version of the beta3 nightly build, which worked.  Last note is to be sure your python sys.path points ONLY to this version.

kmdouglass, 2018-06-21:
@jondaniels 
On Linux, you explicitly set the location of the Python development libraries and the interpreter in the configure script when compiling Micro-Manager. This allows you to build the wrapper for different versions of Python.

```
PYTHON="/usr/bin/python3" ./configure --prefix="/micro-manager/src/build" --with-python="/usr/include/python3.5"
```

Based on this, my guess is that on Windows it's primarily the .props file that needs updating, as well as installing the newer Python interpreter and development libraries on the build machine.

MaximeMaW, 2018-07-31:
See also this thread on the MM mailing list: 
- https://sourceforge.net/p/micro-manager/mailman/message/36380664/
- https://sourceforge.net/p/micro-manager/mailman/message/36380691/

[...]
> One solution would be to compile MM against your desired version of Python;
> I personally have done this with Python 3.6 on Linux and it seems to work.
> On Windows, you need to modify a Visual Studio property file before
> compiling (see the link below for hints).
[...]

Goldensteev, 2018-08-01:
I'm having more troubles than not trying to compile the wrapper with Python 3.6 on Windows, an official transition/compilation would indeed be a great feature!

marktsuchida, 2018-10-02:
Just to let everyone know, Open Imaging is working on this currently...

Goldensteev, 2019-01-23:
Are there any updates on this topic?

hadim, 2019-03-02:
@marktsuchida I don't want to push you but a comment on this feature and also maybe a general word about the status of MM development would be, I think, interesting for everyone. 

For example, I am wondering if you have a date (even an approximative one) for an MM 2 release. I know a vendor who is waiting for it before starting to port its driver to MM 2.

Also, the current development looks kind of scattered to different repo here and on @nicost's repo. Any change to consolidate this? Do you guys sync your repositories between you?

Also, I remember you started to set up a new build system using Gradle. That is pretty exciting as it should lower the "skill required" to install an MM development environment and so wider the potential contributors' list to MM.

Anyway, for plenty of reasons, I think a bunch of people here would love to hear about MM. Even if it's to say that development will keep being slow for a while, at least it's news and it's better than nothing :-)

Wish you and MM all the best!

marktsuchida, 2020-02-21:
Progress towards this is being made in https://github.com/micro-manager/pymmcore/.
It will be a separate install via `pip`.

@hadim Thanks for the kind words, and I apologize for the long blank.

Cc: @oeway

closed 2020-02-21 by marktsuchida