micro-manager/micro-manager #662: Fail to compile MMCoreJ_Wrap using VS2015

SilenceXG, 2018-10-24:
I'm trying to modify and compile mm2's source code. 
Following instructions in https://micro-manager.org/wiki/Building_MM_on_Windows , I use VS2015, Platform Toolset VS2015(v140) Release/x64 config. 3rd party files needed for Release 1.4 from https://valelab4.ucsf.edu/svn/3rdpartypublic/ 

Error Message:
`
Error	MSB6006	"cmd.exe" exited with code 1.	MMCoreJ_wrap	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets	171	
`
Output from console:
1>Target "CustomBuild" in file "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets" from project "E:\mm2-loci\MMCoreJ_wrap\MMCoreJ_wrap.vcxproj" (target "_BuildGenerateSourcesAction" depends on it):
1>       Using "CustomBuild" task from assembly "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.Build.CppTasks.Common.dll".
1>       Task "CustomBuild"
1>         Write Tracking Logs:
1>         	E:\mm2-loci\build\intermediates\Release\x64\MMCoreJ_wrap\MMCoreJ_wrap.tlog\custombuild.write.1.tlog
1>         Read Tracking Logs:
1>         	E:\mm2-loci\build\intermediates\Release\x64\MMCoreJ_wrap\MMCoreJ_wrap.tlog\custombuild.read.1.tlog
1>         No output for E:\MM2-LOCI\MMCOREJ_WRAP\MMCOREJ.I was found in the tracking log; source compilation required.
1>         MMCoreJ.i will be compiled as it was not found in the tracking log.
1>         rem Stage output within IntDir so that it will be cleaned.
1>         set swig_out_dir=E:\mm2-loci\build\intermediates\Release\x64\MMCoreJ_wrap\Swig
1>         set java_src_dir=..\build\intermediates\Swig\mmcorej
1>         rem
1>         echo Removing existing SWIG-generated sources (%swig_out_dir%)...
1>         del /s/q %swig_out_dir% > NUL
1>         echo Creating output directory (%swig_out_dir%)...
1>         mkdir %swig_out_dir% > NUL
1>         echo Running SWIG on E:\mm2-loci\MMCoreJ_wrap\MMCoreJ.i...
1>         E:\mm2-loci\buildscripts\VisualStudio\\..\..\..\3rdpartypublic\swig\swig.exe -c++ -java -package mmcorej -module MMCoreJ -outdir %swig_out_dir% E:\mm2-loci\MMCoreJ_wrap\MMCoreJ.i
1>         rem
1>         echo Removing old copy of Java sources (%java_src_dir%)...
1>         del /s/q %java_src_dir% > NUL
1>         echo Creating directory for Java sources (%java_src_dir%)...
1>         mkdir %java_src_dir% > NUL
1>         echo Copying Java sources to %java_src_dir%...
1>         copy %swig_out_dir%\*.java %java_src_dir%\ > NUL
1>         Running Swig on MMCoreJ.i
1>         Removing existing SWIG-generated sources (E:\mm2-loci\build\intermediates\Release\x64\MMCoreJ_wrap\Swig)...
1>         Creating output directory (E:\mm2-loci\build\intermediates\Release\x64\MMCoreJ_wrap\Swig)...
1>         A subdirectory or file E:\mm2-loci\build\intermediates\Release\x64\MMCoreJ_wrap\Swig already exists.
1>         Running SWIG on E:\mm2-loci\MMCoreJ_wrap\MMCoreJ.i...
1>         The system cannot find the path specified.
1>         Removing old copy of Java sources (..\build\intermediates\Swig\mmcorej)...
1>         Creating directory for Java sources (..\build\intermediates\Swig\mmcorej)...
1>         A subdirectory or file ..\build\intermediates\Swig\mmcorej already exists.
1>         Copying Java sources to ..\build\intermediates\Swig\mmcorej...
1>         The system cannot find the file specified.
1>     1>
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1.
1>       Done executing task "CustomBuild" -- FAILED.
1>     1>
1>Done building target "CustomBuild" in project "MMCoreJ_wrap.vcxproj" -- FAILED.
1>
1>Build FAILED.

nicost, 2018-10-24:
The instruction say: "Micro-Manager's C++ components (Core, MMCoreJ, MMCorePy, and device adapters) are currently built using Visual Studio 2010 Express."  

PRs for a full build with newer VS versions are welcome.  Please close this issue.

SilenceXG, 2018-10-24:
Thank you for your reply.

closed 2018-10-24 by SilenceXG

hr-yang, 2024-04-24:
I am meeting the same problem with VS2019(V142).   JAVA:1.8.0_412. ANT:1.10.12 following instructions in https://micro-manager.org/wiki/Building_MM_on_Windows 
[image]

marktsuchida, 2024-06-13:
@hr-yang Your error is different from the one described in this issue. Please check that you have 3rdpartypublic in the required location. If you still have this problem, please file a new issue (and provide some more details on what you did).

(MSB8066: Custom build for 'item-list' exited with code 'error-code')