micro-manager/micro-manager #666: I want to write a micro-manager plugin for ImagePy

yxdragon, 2019-01-09:
Hi, I am building a image process framework [ImagePy](https://github.com/Image-Py/imagepy).It is a framework like imagej, but implemented in pure python. 

I am going to build a micro-manager plugin, like the imagej one. But some trouble below:
1. It seems that there is no easy way to access mmcore through python3. [](https://github.com/micro-manager/micro-manager/issues/633)
2. I found the java api document, but about the python's, I just found a demo in official site. Is there some detail document?

I can write all the interactive ui. I want to know if you can give a easy solution for python3(win and linux), and help to write some document in detail (If there is not now). Thanks!

I need your help, and expect cooperation.

nicost, 2019-01-10:
The Micro-Manager mailing list may be a better place for this discussion, as there are actually people using Python 3 with Micro-Manager there. Python 3 support in Micro-Manager was discussed there before: http://micro-manager.3463995.n2.nabble.com/ (search for Python). It seems that your best bet is to ask Open Imaging what the status of a Python 3 port is.

However, a lot of the Micro-Manager's application logic is contained in the Java layer (for instance, the "Acquisition Engine" that runs multi-dimensional acquisitions is in the Java layer, file reading and saving is in Java, etc..), and there is much more to making this work than just writing a GUI.  Nevertheless, simple functionality in a python GUI would be great, and I don't want to discourage you.  

Lastly, I guess that you are using Google translate.  Saying "I expect cooperation" sounds threatening, and I don't think that is what you meant.

yxdragon, 2019-01-10:
Thanks, I would search the mailing list, In fact, I have succeed, (install 2.0, and compile with visual studio, then replace the python wrapper), But I think It is different between "it is possible" and "it is easy to do with any version python anywhere", right?

In fact, I do not known microscope well, But in my opinion, there is no big difference from operate a directX Show camera, just set some properties, then got the image data. And If I have a useful api and document, I can do it in one or two days. (may be I am a little arrogantly)

I am sorry for the "I expect cooperation", I did not use Google translate, but I am not English native. I means that I wish we can cooperate to build the program better.

nicost, 2019-01-10:
Thing is, it is not just about getting the image data, it is about executing sequences of hardware events (moving a stage, moving a filter in a filter wheel, opening a shutter, starting exposure, ending exposure, closing the shutter, reading out the camera, saving the image data with metadata in a workable/maintainable file format).  Most of that logic resides in the Java layer, and will be a lot of work (i.e., one or two years rather than days;) to replicate.  Nevertheless, just being able to grab an image from the camera, will be a nice start (it will not make it usable for most biologists, but hey, better to start somewhere than nowhere at all).

It would be nicest and easiest for you if the Micro-Manager distribution would come with Python3 wrappers.  That is out of my hands, hopefully things will change within half a year or so (and in the mean time you could Python3 wrappers for a specific MM version available yourself).

""Collaborate" is better than "cooperate".  "Cooperate" kind of signals "force to collaborate".

yxdragon, 2019-01-11:
May be there is lot of work to do, but I think that is what the api level should do, not the app level. I am not aware of micoscopy, but I have experience to write many directX show camera software, oscilloscope ctrl software, and many adc software. The all work is, read/write register, (set filter, mode, time base, voltage range ...), then get data one time. Or in continue mode, set some trigger, and give some call back function. I can not even conceive of that I need one or two years to write an app with an api. In your words, I think it would be less usable for programmer. Do you have plan about making it easy to use?

nicost, 2019-01-11:
That part of the API is all in Java (see: https://micro-manager.org/wiki/Micro-Manager%20Project%20Overview), so if you find a way to interface to the Java-based API, then you are good to go.  Again, a simple GUI in Python on top of the Micro-Manager C++ API is useful, but will lack most of the relevant functionality (which is in the Java layer).  I am not aware of anyone with plans to port that functionality to another language (and I certainly will not do that myself). In the mean time, the code (both the C++ and Java layer) is still very useful for a lot programmers and microscopists.

yxdragon, 2019-01-12:
OK, Thanks!

closed 2019-01-12 by yxdragon