micro-manager/micro-manager #671: Failing to build OpenSUSE

wsphillips, 2019-04-12:
Throwing errors while trying to build devices. All core modules build fine. Error files attached. Any advice? clang++ is using gcc v7.2 Java 1.8. 
[error.txt](https://github.com/micro-manager/micro-manager/files/3074715/error.txt)
[error.txt](https://github.com/micro-manager/micro-manager/files/3074717/error.txt)

I even tried the very nice script linked in #573 

Is there a way to safely build with only a couple devices? Looks like theres some dependencies between devices? 

Specifically, I only NEED PVCAM, ZeissCAN29, and Axiocam. Which other devices are those dependent on? How can I comment out the others from building so to not spend time solving build errors for something I wont use?

nicost, 2019-04-12:
Which version of the code are you building?  The current code for the HamiltonMVP has "result = std::atoi(s.c_str());" on the line where you get the error, and you have "result = atoi(s.c_str()); " there.  I have the feeling that you are building 2.0-beta.  2.0-beta is currently not (or sparsely) maintained.  Use 1.4 (in the svn-mirror branch in this repo) or better, 2.0-gamma (https://github.com/nicost/micro-manager/tree/ViewerPlusCV).

If you really want to get beta to work, edit DeviceApaters/Makefile.am to remove the device adapters that you do not like.  B.t.w, I doubt that PVCAM will build on Linux and the AxioCam adapter most certainly will not work on Linux (there are no Linux drivers for these, and the device adapters are merely the "glue" between the Micro-Manager core and the drivers).

wsphillips, 2019-04-13:
ah yeah. Axiocam isn't critical--thats just a test camera. For PVCAM--I just so happen to be trialling brand new linux drivers for photometrics ;) Any advice? I am supposed to be sent their new SDK this week, and I am assuming the headers are more or less the same (if not identical). Looking briefly, I can see the PVCAM device adapter had some code to look for the ancient legacy linux drivers when on linux--should that be enough to bootstrap the current api? (i.e. looking to run a Prime 95B as I do on Windows boxes)

For the build--yes I was cloning into the 2.0-beta branch. I was a bit confused, since I had seen the gamma binaries. Thanks for setting me straight. I will confirm it builds properly once I give it a proper go with your fork.

wsphillips, 2019-04-13:
@nicost I managed to get a successful build using 2.0-gamma, however I had to comment out one device that was giving errors (WieneckeSinske). I am super sure I won't be using a piezo stage for inverted scopes, so safe to say "up and running" now. 

I may come back asking about PVCAM with these new linux binaries however!

closed 2019-04-13 by wsphillips

nicost, 2019-04-14:
If you show/send me the error you get with the WieneckeSinske device adapter, I'll see if there is an easy fix (usually these things concern names/locations of header files).  Always nice if everything just compiles;)