micro-manager/micro-manager #672: IDS_uEye.h:53:10: fatal error: uEye.h: No such file or directory SOLVED

lindseypeng, 2019-04-16:
trying to install micromanager on ubuntu18
i already had ids uEye installed for my camera and i suspect that caused some problem with micromanager installation
i tried clean install a 4 times and it keep throwing me the same error of missing uEye.h file.
I found the uEye.h file from previous uEye driver installation and moved it into ~/mm/micro-manager1.4/DeviceAdaptors/IDS_uEye and it works

PCCV, 2019-06-14:
Under linux, the file is ueye.h (note undercase)
I corrected this with 
`sed -i "s/<uEye.h>/<ueye.h>/g"  DeviceAdapters/IDS_uEye/IDS_uEye.h`

nicost, 2019-06-14:
But is it called the same on other platforms?  I have no idea, so used #ifdef linux preprocessor directives.  Please test MM 1.4 >= commit r17077 and let us know if this now works for you.

DanielArnett, 2019-12-13:
This came up as a Google search result,
> Under linux, the file is ueye.h (note undercase)

Thanks for catching this @PCCV !

closed 2020-01-31 by nicost