micro-manager/micro-manager #677: Image sequences within runnables (worked in 1.4)

ctr26, 2019-08-15:
Hi all,

I wrote a beanshell script for 1.4 which was able to run a runnable on each frame of a MDA acquisition and within each frame create an image montage of 9 images.

If I attempt to run an image sequence during a runnable I get an error that claims my camera has crashed in MM. Is this a feature change? Is there anyway I can make a similar runnable to the one I wrote before?

Best

Craig

nicost, 2019-08-15:
Which version does this pertain to?  For 2.0-gamma, please post on https://github.com/nicost/micro-manager.  2.0-beta (which lives here) is no longer maintained.  

Assuming 2.0-gamma, what you see is most likely a side-effect (not something the code is regularly tested for;).  Can you post an example script (preferably something that replicates the issue with the Demo configuration?

Also, I can not close issues here, can you do so and move to 2.0-gamma?

ctr26, 2019-08-15:
I'm very confused by this. MM here is no longer being developed but it has newer commits than gamma?

nicost, 2019-08-15:
It is highly confusing!  However, latest commits in micro-manager/micro-manager branches:

master: July 19, 2018
mm2 (used to build 2.0-beta): Nov3, 2018
svn-mirror (automatically pulling commits from svn 1.4): 11 days ago.

nicost/micro-manager:

ViewerPlusCV (used to build 2.0-gamma):16hrs ago.

So, not sure why you conclude that there are newer commits here than in gamma.

I don't have permissions to set things straight on micro-manager/micro-manager....

closed 2019-12-07 by nicost