micro-manager/micro-manager #681: Compile Error: InfoMessage4 << "test" <<com

JiangXL, 2019-12-13:
Micromanager branch: svn-mirrors
System: Linux deepin 15.11(based on Debian)

I have to commit `Oxxius_combiner.cpp:329` to continue compile.

``` bash
Oxxius_combiner.cpp:329:27: error: no match for ‘operator<<’ (operand types are ‘std::basic_ostream<char>’ and ‘std::ostringstream {aka std::__cxx11::basic_ostringstream<char>}’)
    InfoMessage4 << "test" << com;
    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
Oxxius_combiner.cpp:329:27: note: candidate: operator<<(int, int) <built-in>
Oxxius_combiner.cpp:329:27: note:   no known conversion for argument 2 from ‘std::ostringstream {aka std::__cxx11::basic_ostringstream<char>}’ to ‘int’
In file included from /usr/include/c++/6/istream:39:0,
                 from /usr/include/c++/6/sstream:38,
                 from ../../MMDevice/ImageMetadata.h:37,
                 from ../../MMDevice/MMDevice.h:49,
                 from Oxxius_combiner.h:15,
                 from Oxxius_combiner.cpp:14:
```

marktsuchida, 2020-02-12:
My first guess is a missing `#include <sstream>`.

tlnagy, 2020-02-13:
I'm also running into a similar error when compiling micro-manager from source on an Kubuntu 19.10 machine:

```
/bin/bash ../libtool  --tag=CXX   --mode=compile g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" -DPACKAGE_STRING=\"Micro-Manager\ 1.4\" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_BOOST=/\*\*/ -DHAVE_BOOST_THREAD=/\*\*/ -DHAVE_BOOST_ASIO=/\*\*/ -DHAVE_BOOST_SYSTEM=/\*\*/ -DHAVE_BOOST_CHRONO=/\*\*/ -DHAVE_BOOST_DATE_TIME=/\*\*/ -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I.    -I/home/tlnagy/Documents/micro-manager/DeviceAdapters/../MMDevice -pthread -I/usr/include -g -O2 -MT Oxxius_combiner.lo -MD -MP -MF .deps/Oxxius_combiner.Tpo -c -o Oxxius_combiner.lo Oxxius_combiner.cpp
libtool: compile:  g++ -DPACKAGE_NAME=\"Micro-Manager\" -DPACKAGE_TARNAME=\"micro-manager\" -DPACKAGE_VERSION=\"1.4\" "-DPACKAGE_STRING=\"Micro-Manager 1.4\"" -DPACKAGE_BUGREPORT=\"info@micro-manager.org\" -DPACKAGE_URL=\"\" -DPACKAGE=\"micro-manager\" -DVERSION=\"1.4\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" "-DHAVE_BOOST=/**/" "-DHAVE_BOOST_THREAD=/**/" "-DHAVE_BOOST_ASIO=/**/" "-DHAVE_BOOST_SYSTEM=/**/" "-DHAVE_BOOST_CHRONO=/**/" "-DHAVE_BOOST_DATE_TIME=/**/" -DHAVE__BOOL=1 -DHAVE_STDBOOL_H=1 -DSTDC_HEADERS=1 -DHAVE_MEMSET=1 -I. -I/home/tlnagy/Documents/micro-manager/DeviceAdapters/../MMDevice -pthread -I/usr/include -g -O2 -MT Oxxius_combiner.lo -MD -MP -MF .deps/Oxxius_combiner.Tpo -c Oxxius_combiner.cpp  -fPIC -DPIC -o .libs/Oxxius_combiner.o
In file included from ../../MMDevice/MMDevice.h:49,
                 from Oxxius_combiner.h:15,
                 from Oxxius_combiner.cpp:14:
../../MMDevice/ImageMetadata.h:308:58: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
  308 |    MetadataSingleTag GetSingleTag(const char* key) const throw (MetadataKeyError)
      |                                                          ^~~~~
../../MMDevice/ImageMetadata.h:315:56: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
  315 |    MetadataArrayTag GetArrayTag(const char* key) const throw (MetadataKeyError)
      |                                                        ^~~~~
Oxxius_combiner.cpp: In member function ‘virtual int OxxiusCombinerHub::DetectInstalledDevices()’:
Oxxius_combiner.cpp:329:27: error: no match for ‘operator<<’ (operand types are ‘std::basic_ostream<char>’ and ‘std::ostringstream’ {aka ‘std::__cxx11::basic_ostringstream<char>’})
  329 |    InfoMessage4 << "test" << com;
      |    ~~~~~~~~~~~~~~~~~~~~~~ ^~ ~~~
      |                 |            |
      |                 |            std::ostringstream {aka std::__cxx11::basic_ostringstream<char>}
      |                 std::basic_ostream<char>
Oxxius_combiner.cpp:329:27: note: candidate: ‘operator<<(int, int)’ <built-in>
  329 |    InfoMessage4 << "test" << com;
      |    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
Oxxius_combiner.cpp:329:27: note:   no known conversion for argument 2 from ‘std::ostringstream’ {aka ‘std::__cxx11::basic_ostringstream<char>’} to ‘int’
In file included from /usr/include/c++/9/istream:39,
                 from /usr/include/c++/9/sstream:38,
                 from ../../MMDevice/ImageMetadata.h:37,
                 from ../../MMDevice/MMDevice.h:49,
                 from Oxxius_combiner.h:15,
                 from Oxxius_combiner.cpp:14:
/usr/include/c++/9/ostream:108:7: note: candidate: ‘std::basic_ostream<_CharT, _Traits>::__ostream_type& std::basic_ostream<_CharT, _Traits>::operator<<(std::basic_ostream<_CharT, _Traits>::__ostream_type& (*)(std::basic_ostream<_CharT, _Traits>::__ostream_type&)) [with _CharT = char; _Traits = std::char_traits<char>; std::basic_ostream<_CharT, _Traits>::__ostream_type = std::basic_ostream<char>]’
  108 |       operator<<(__ostream_type& (*__pf)(__ostream_type&))
      |       ^~~~~~~~
/usr/include/c++/9/ostream:108:36: note:   no known conversion for argument 1 from ‘std::ostringstream’ {aka ‘std::__cxx11::basic_ostringstream<char>’} to ‘std::basic_ostream<char>::__ostream_type& (*)(std::basic_ostream<char>::__ostream_type&)’ {aka ‘std::basic_ostream<char>& (*)(std::basic_ostream<char>&)’}
  108 |       operator<<(__ostream_type& (*__pf)(__ostream_type&))
      |                  ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/ostream:117:7: note: candidate: ‘std::basic_ostream<_CharT, _Traits>::__ostream_type& std::basic_ostream<_CharT, _Traits>::operator<<(std::basic_ostream<_CharT, _Traits>::__ios_type& (*)(std::basic_ostream<_CharT, _Traits>::__ios_type&)) [with _CharT = char; _Traits = std::char_traits<char>; std::basic_ostream<_CharT, _Traits>::__ostream_type = std::basic_ostream<char>; std::basic_ostream<_CharT, _Traits>::__ios_type = std::basic_ios<char>]’
  117 |       operator<<(__ios_type& (*__pf)(__ios_type&))
      |       ^~~~~~~~
[... cut ...]
  363 |     operator<<(basic_ostream<_CharT, _Traits>& __os, _Put_time<_CharT> __f)
      |     ^~~~~~~~
/usr/include/c++/9/iomanip:363:5: note:   template argument deduction/substitution failed:
Oxxius_combiner.cpp:329:30: note:   ‘std::__cxx11::basic_ostringstream<char>’ is not derived from ‘std::_Put_time<_CharT>’
  329 |    InfoMessage4 << "test" << com;
      |                              ^~~
make: *** [Makefile:538: Oxxius_combiner.lo] Error 1

```

Adding `#include <sstream>` to the `.cpp` didn't seem to resolve the problem

JiangXL, 2020-02-13:
@tlnagy Hi, it seem that recent Boost does not work well with Micro-manager. Actually, I use the [bash script](https://github.com/hadim/mm_scripts/blob/master/build-mm.sh) by @hadim , which work smoothly in my machine with little modication.

tlnagy, 2020-02-15:
Thanks! That almost gets me there, I had to uncomment `make fetchdeps` since that command is missing from the generated `Makefile`, but I don't get a micro-manager executable at the end?

`make install` ends on the following:
```
make[3]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters/pgFocus'
make[2]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters/pgFocus'
make[2]: Entering directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters'
make[3]: Entering directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters'
make[3]: Nothing to be done for 'install-exec-am'.
make[3]: Nothing to be done for 'install-data-am'.
make[3]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters'
make[2]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters'
make[1]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager/DeviceAdapters'
make[1]: Entering directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make[2]: Entering directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make  install-exec-hook
make[3]: Entering directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make[3]: Nothing to be done for 'install-exec-hook'.
make[3]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make  install-data-hook
make[3]: Entering directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make[3]: Nothing to be done for 'install-data-hook'.
make[3]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make[2]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager'
make[1]: Leaving directory '/home/tlnagy/Documents/micro-manager/micro-manager'
```

JiangXL, 2020-02-15:
I also has problem with `make fetchdeps`:
``` bash
mm.fetchdeps.download-ivy:
      [get] Getting: http://repo2.maven.org/maven2/org/apache/ivy/ivy/2.3.0/ivy-2.3.0.jar
      [get] To: /home/qblab/Bin/mm-sh/micro-manager/dependencies/ivy/ivy-2.3.0.jar
      [get] Error getting http://repo2.maven.org/maven2/org/apache/ivy/ivy/2.3.0/ivy-2.3.0.jar to /home/qblab/Bin/mm-sh/micro-manager/dependencies/ivy/ivy-2.3.0.jar
BUILD FAILED
/home/qblab/Bin/mm-sh/micro-manager/buildscripts/fetchdeps.xml:12: java.net.UnknownHostException: repo2.maven.org
```
I think it is the same of #683 

But I'm confused the branch of micromanager. I can't find the laest version, which is used for building mm2-gamma. @marktsuchida may have suggestion.

JiangXL, 2020-02-15:
After `make install`, the executable files should be found in `Fiji.app/mmimagej` instead of `micromanager` folder. And I launch mm by execute `./Fiji.app/mmimagej` or launch Fiji.app then boot from Fiji pulgin.

nicost, 2020-02-16:
> ```shell
> mm.fetchdeps.download-ivy:
>       [get] Getting: http://repo2.maven.org/maven2/org/apache/ivy/ivy/2.3.0/ivy-2.3.0.jar
> ```

Which branch are you using?  Currently the link in master will be https://repo1.maven.org and work.

> But I'm confused the branch of micromanager. I can't find the laest version, which is used for building mm2-gamma. @marktsuchida may have suggestion.

mm2-gamma is build from the master branch here (as it says in the README on the front page here in this github repo).

JiangXL, 2020-02-16:
At that point, I used mm2 branch. Now I switch to master(68f3383) , no connection error again. Sorry for this noise due to my carelessness. 

But `Compile Error: InfoMessage4 << "test" <<com` still occur.

tomhanak, 2021-03-10:
The `com` is `std::ostringstream`, i.e. an output stream one can write into, but nobody can read from it directly. That's what the error says.
This problem is already fixed in git (ecfe0cce) by sending `com.str()` to `InfoMessage4` stream. It was not backported to svn yet.

This issue can be closed.

closed 2021-03-10 by nicost