micro-manager/micro-manager #683: Problem with ivy repo

ckemere, 2020-01-24:
I got the following error:

`      [get] Error opening connection java.io.IOException: Server returned HTTP response code: 501 for URL: http://repo2.maven.org/maven2/org/apache/ivy/ivy/2.3.0/ivy-2.3.0.jar
`

I resolved it by swapping `http` to `https` (maybe unnecessary) and `repo2` to `repo1` in `fetchdeps.xml`.

nicost, 2020-01-24:
Which branch?  Did you pull to get the latest version of the code?

ckemere, 2020-01-24:
This was in a fresh pull from this repository (so the master branch). The 3rd party repository was through SVN

`svn co https://valelab4.ucsf.edu/svn/3rdpartypublic/`

nicost, 2020-01-24:
master branch is not in use.  It will be resurrected in the future.  You can use the svn-mirror branch (for 1.4), or nicost/micro-manager ViewerPlusCV branch (for 2.0-gamma, soon to become the official 2.0 here).

closed 2020-01-24 by nicost